sstsimulator/sst-elements #16: Trac: #297 MemHierarchy support for FLUSH cpu instruction

nmhamster, 2015-10-07:
MH should support flushing. This functionality is required by some OSesCPUs An L1 should look at all its cache lines and writeback any valid ones. Potential issues: A) The cache line is in transient state. Cache should probably wait for response before writing back B) Cache should temporarily disable other incoming requests (prefetcher, cpu) while flushing C) Assert that no cache lines have a LOCK bit D) How many writebacks to send per cycle Gotta be careful not for overflowdeadlock situations

allevin, 2015-10-08:
Moved to SST v6.0 but discussion underway for how to implement. Possibly need for non-volatile flushing and SST checkpointing studies.

nmhamster, 2017-04-16:
Initial support now included in MemHierarchy for SST v7.0 release.

closed 2017-04-16 by nmhamster