sstsimulator/sst-elements #23: Trac: #347 Zero queue fill in Ariel on forking children

nmhamster, 2015-10-07:
When the application forks, the simulation crashes as Ariel core 0s instruction queue becomes empty. The error message is pasted below: sst: ........sstelementsarielarielcore.cc:236: bool SST::ArielComponent::ArielCore::refillQueue(): Assertion 0 failed.

allevin, 2015-10-08:
This occurs on machines which do not correctly support fork on compute nodes (we see this on the Cray XC30 test machine) or when the application being analyzed calls fork and PIN does not follow the child. Need to investigate further but we have already added PIN parameter for request following of fork calls and bug still remains.

allevin, 2015-10-08:
Moving to SST v6.0. Still investigating whether this is a common use case.

nmhamster, 2016-05-30:
Fixed, Ariel running successfully on Cray XC platform.

closed 2016-05-30 by nmhamster