sstsimulator/sst-elements #25: Trac: #359 Isolated failure of the prospero tests on Ubuntu 14.04

nmhamster, 2015-10-07:
January 6th, the nightly test on Ubuntu 14.04 (#181) that included prospero failed. The three prospero tests that use DramSim passed, but the three that dont use DramSim failed. There were no SVN changes from the previous that passed and there has been no recurrence. There have been no related failures on other hosts including a new Ubuntu 14.04 VM. The trace-back from the failure is attached. (Could not collect gdb into because the failure was a one-only.) Running the test under Valgrind on sst-test is completely clean. Running Valgrind on Ubuntu 14.04 is a story unto itself. Valgrind has several complaints about using uninitialized variables. The entire trace-back is in Python library routines. Ubuntu 14.04 uses Python 2.7. CentOS and most other place we run use Python 2.6

allevin, 2015-10-08:
Attachment Prospero.fail​ added

trace-back of Prospero fail January 6, 2015

allevin, 2015-10-08:
On the Valgrind + Python + Ubuntu14.04 front:
It is know that Python and Valgrind don't really play well together. Try adding:

```
--suppressions=/usr/lib/valgrind/python.supp
```

To the valgrind command line. That should enable the Python-associated suppressions.

allevin, 2015-10-08:
Last recorded occurrence February 3rd, 2015.
(July 9, 2015) Search finds none in last month.

allevin, 2015-10-08:
[Prospero.fail.txt](https://github.com/sstsimulator/sst-elements/files/10177/Prospero.fail.txt)

closed 2017-04-16 by nmhamster