sstsimulator/sst-elements #3: Trac: #54 Power modeling with sst?

nmhamster, 2015-10-07:
Reported by gortipavan, Mar 13, 2013 What steps will reproduce the problem 1. Installing sst-2.3.0 stable and sst-gem5-2.3.0, with Hotspot, and McPat 2. Simulation using example_power.xml file 3. What is the expected output What do you see instead I expect power statistics to be output for executing an application. I do not see them. What version of the product are you using On what operating system (provide version) Are you developing using a Virtual Machine environment If so, provide both virtual and actual machine OS names and versions. sst-2.3.0 sst-gem5-2.3.0 Red Hat Linux 5.8 Tikanga Please provide any additional information below. When building sst, I get the message that gem5 power models are disabled for this build. Is this a problem If so, how to get around this

allevin, 2015-10-07:
Initial support for statistics in components is now available in SST v5.0 onwards. This will now be merged with a map interface for components onto McPAT energy models. This work is currently in development.

allevin, 2020-01-29:
Not a currerent priority

closed 2020-01-29 by allevin