sstsimulator/sst-elements #33: Trac: #405 Occasional errors seen processing XML input file to SST on Yosemite

nmhamster, 2015-10-07:
This occurs running the Patterns test Suite. This presumably results from the Python version upgrade on Yosemite. It is not a solid (reproducible) failure. To date it has only been seen on tests of the 5.0 release and the 5.0.1 pre-release. The XML file is attached and is generated in the testSuite. The xml is 96,000 lines (3.7 megabytes) It was not previously replaced with a python input file because it is generated but more importantly, if one uses SST to auto-convert from xml to python, the file grows by a factor of 25! Log file of the failure is also attached

allevin, 2015-10-08:
Log file of occasionly SST failure processing xml input file

allevin, 2015-10-08:
Related Issues: #311 and #363

allevin, 2015-10-08:
previous comment lost.
SORRY The XML is not attached. It significantly exceeds the 1/4 megabyte limit for attachments.
Contact me at ​jpvandy@sandia.gov for access to the file.

allevin, 2015-10-08:
[Stray-Yosemite.txt](https://github.com/sstsimulator/sst-elements/files/10219/Stray-Yosemite.txt)

jpvandy, 2016-12-13:
Patterns is no longer in the release.

closed 2016-12-13 by jpvandy