sstsimulator/sst-elements #4: Trac: #85 For mcopterian the result on MacOS is not deterministic

nmhamster, 2015-10-07:
The results for test2 from SVN from repeated runs produce different answers on MacOS. It has been verified multiple times that the random number streams used is NOT the issue. Repeated runs with the same initial seed yield multiple answers. (This is not observed on any of the Linux platforms regularly tested.)

allevin, 2015-10-07:
Not fixed for v4.1.0. Jeanine is investigating and plans to provide fix for SST v5.0.0

allevin, 2015-10-07:
While running Valgrind on various SST tests noted that:
(1) McOpteron? test1 and test2 have different characteristic from Valgrind.
(2) Valgrind (on CentOS) finds 664,707 errors from 11 contexts on McOpteron?.

allevin, 2015-10-07:
Not to be included in SST v5.0

allevin, 2015-10-07:
Test #2 is now failing on Linux.
Am going to universally disable test2.
Do not know when the symptom on Linux appeared. The SST run is producing a grossly different output file reporting many error. However the test Suite was also broken and consequently, the test failure was not being reported to Jenkins.

allevin, 2015-10-07:
Probably of no interest, but the mcopteron test failure appears to be correlated with mod set #9557

allevin, 2015-10-07:
Jeanine, any update on McOpteron/McNiagara? models?

nmhamster, 2016-07-05:
Moved to attic, shelved for future.

closed 2016-07-05 by nmhamster