sstsimulator/sst-elements #5: Trac: #137 Prefetcher additions slow down VaultSim test

nmhamster, 2015-10-07:
What steps will reproduce the problem Compare the testSuite_VaultSim.sh results before and after the repository changes of approximately December 3rd, 2013 (rev 6024-6031). After that date, all VaultSim tests fail because of the changed output. Reference file was not updated, because the resulting change was not desired.

allevin, 2015-10-07:
January 10th: Will update the reference file to allow nightly tests to better monitor other issue. This does not fixed Prefetching vs. VaultSim?, but does make the test accept the result.
Jan 10, 2014
Project Member #1 jpvandy4sst
Minor Addition: It is only the sdl3 case that is failing. sdl1 and sdl2 pass.

nmhamster, 2019-03-08:
Prefetching and VaultSim working correctly in tests. Closing issue, re-open if needed.

closed 2019-03-08 by nmhamster