sstsimulator/sst-elements #34: Trac: #381 Less counter intuitive variable names in loadInfo.py

nmhamster, 2015-10-08:
In sstelementsembertestloadInfo.py (around line 100) In function initWork, each work contained in list workList is enumerated and passed to function readWorkList In that function, the work is renamed workList and in another for loop, each work of this workList (previously work) is processed. These multiple work, worklist, workflow, etc. makes loadInfo.py extremely hard to read. Perhaps other variable names could be employed Also, for what does nidList stand for And what if the different work contained in the worklist as defined in function readWorkList define through their cmd different nidList Same question for ranksPerNode

allevin, 2020-01-29:
@feldergast will address.

hughes-c, 2023-10-26:
Won't fix.

closed 2023-10-26 by hughes-c