sstsimulator/sst-elements #41: Trac: #443 CHDL component is not ready for release.

nmhamster, 2015-10-08:
Previous issue #329. It will not build with gcc-4.4.7 because of C++11 usage - this is probably not to get fixed. (Nightly testing is disabled if C++11 is not available.) Current version of chdl will not build with Clang compiler. Appears to be fine with gcc-4.7 and gcc-4.8.1. The log of a Clang compile failure is attached. (Nightly testing is disabled on Clang compilers.) The test needs two input files which are being fetched from kersey.com, which need to be install in sst downloads. The external component for chdl comes from three git downloads which are not from a release but from a HEAD of a repository.

allevin, 2015-10-08:
Log of CHDL external build failure

allevin, 2015-10-08:
gcc-4.4.7 support is a still a requirement for SST 5.1.0. (move this issue to Milestone 6.0)

allevin, 2015-10-08:
[CHDL.vs.Clang.txt](https://github.com/sstsimulator/sst-elements/files/10235/CHDL.vs.Clang.txt)

closed 2016-06-29 by jwilso