sstsimulator/sst-elements #43: Trac: #458 MacSim tests fail on Static Yosemite

nmhamster, 2015-10-08:
The problem is only seen on Static Yosemite. Valgrind (on CentOS) reports over 2 million errors in 49 contexts from MacSim, but no segfault on CentOS. Its a segment fault. [yosemitexc63:66125] create_issues.sh create_issues.sh~ create_issues_orig.sh csv csvprocessor.c report_12_4.csv run_core_issues_final.sh run_create_issues.sh run_create_issues_orig.sh run_create_milestones.sh run_create_milestones.sh~ run_elements_issues_final.sh run_workbench_issues_final.sh Process received signal create_issues.sh create_issues.sh~ create_issues_orig.sh csv csvprocessor.c report_12_4.csv run_core_issues_final.sh run_create_issues.sh run_create_issues_orig.sh run_create_milestones.sh run_create_milestones.sh~ run_elements_issues_final.sh run_workbench_issues_final.sh [yosemitexc63:66125] Signal: Segmentation fault: 11 (11) [yosemitexc63:66125] Signal code: Address not mapped (1) [yosemitexc63:66125] Failing at address: 0xb [yosemitexc63:66125] [ 0] 0 libsystem_platform.dylib 0x00007fff9009ff1a _sigtramp + 26 [yosemitexc63:66125] [ 1] 0 0x00007fff5dc42a10 0x0 + 140734766524944 [yosemitexc63:66125] [ 2] 0 sstsim.x 0x00000001021c4073 _ZN6core_c11run_a_cycleEb + 163 [yosemitexc63:66125] [ 3] 0 sstsim.x 0x00000001021e4b4a _ZN8macsim_c11run_a_cycleEv + 986 [yosemitexc63:66125] [ 4] 0 sstsim.x 0x0000000102158358 _ZN3SST6MacSim15macsimComponent11ticReceivedEy + 184 [yosemitexc63:66125] [ 5] 0 sstsim.x 0x0000000101fd996c _ZN3SST5Clock7executeEv + 76 [yosemitexc63:66125] [ 6] 0 sstsim.x 0x0000000102033790 _ZN3SST10Simulation3runEv + 400 [yosemitexc63:66125] [ 7] 0 sstsim.x 0x0000000101fbc0f2 main + 8658 [yosemitexc63:66125] [ 8] 0 libdyld.dylib 0x00007fff876f85c9 start + 1 [yosemitexc63:66125] create_issues.sh create_issues.sh~ create_issues_orig.sh csv csvprocessor.c report_12_4.csv run_core_issues_final.sh run_create_issues.sh run_create_issues_orig.sh run_create_milestones.sh run_create_milestones.sh~ run_elements_issues_final.sh run_workbench_issues_final.sh End of error message ***

jpvandy, 2016-01-26:
MacSim test currntly fails with page fault on all MacOS tests.

Disabling the test.

nmhamster, 2016-03-14:
@hyesoon @hyojongk - any update on Mac OS X support for MacSim? Thank you!

ramyadhadidi, 2016-05-25:
@nmhamster I recently tested macsim with OS X El Capitan (10.11.5) without any error. I used our devel repository (https://github.com/gthparch/macsim_devel). How should we debug this issue then? I gave you access to our macsim_devel repo.

jpvandy, 2016-10-04:
We have discontinued attempting tests with the Static Option.

closed 2016-10-04 by jpvandy