sstsimulator/sst-elements #44: Trac: #464 VaultSimC Initialization Fails

nmhamster, 2015-10-08:
The following message is produced many times when trying to initialize VaultSimC using chdlComponents initial memory contents configuration parameter: Memory received unexpected Init Command: 2 These start as writes. Elsewhere (memhierarchys memcontroller), command 2 (GetX) is used to provide init data. Even if we allow GetX commands in addition to writebacks to be used to provide initial contents, initialization seems to fail; the memory contents cannot be read back.

allevin, 2017-03-24:
@afrodri Not sure if this is still an issue.  Note: CHDL is now in the elements-archive.  If not a problem can you close issue?

allevin, 2020-04-03:
@afrodri is this still an issue?

closed 2020-09-30 by allevin