sstsimulator/sst-elements #64: Merlin Dragonfly Question

shavvn, 2015-12-08:
Hi I was playing with merlin and trying to build a dragonfly network. In the /merlin/topology/dragonfly.cc source code I found (line 215):

```
uint32_t topo_dragonfly::router_to_group(uint32_t group) const
{
    /* For now, assume only 1 connection to each group */
```

Does it mean I cannot have more than 1 inter-group link per router?

Thanks,
Shang

feldergast, 2015-12-08:
That is currently correct.  I'm working this week on removing that constraint.

Scott

shavvn, 2015-12-08:
great  good to know that! thanks!

nmhamster, 2016-02-12:
Scott, I think you have fixed this now correct?

nmhamster, 2016-02-28:
This is confirmed fixed in recent additions from @feldergast. Closing issue. Code is now in the devel branch.

closed 2016-02-28 by nmhamster