sstsimulator/sst-elements #74: SST-MacSim is Not Compatible with SST Threaded PDES Core

nmhamster, 2016-01-18:
Fails on Linux with n > 1 for thread count.

nmhamster, 2016-03-14:
@hyesoon @hyojongk - any update on testing of MacSim with the SST Threaded PDES core at all? Thank you!

jpvandy, 2016-03-30:
For n>1, on Linux, test runs to completion, but the output is different than expected

hyesoon, 2016-04-06:
Yes we are also able to reproduce the failed cases for n > 2 .  The outputs are slightly different. 
I'm not sure exactly how PDE works. if PDE somehow allows parallel execution of macsim cores, or memhierarchy (our test cases are built on memHieararchy),  this can slightly changes the simulation time.  We could relax the script to allow some simulation cycle differences for now to make the test success even with n > 2. 
Can we provide some point to look for the PDE related code in SST-core?

feldergast, 2016-04-06:
The PDES core is designed to allow reproducible results between serial and parallel jobs.  There are two primary cases that can get you into trouble with parallel simulations.  The first is use of static variables.  For MPI, you end up with multiple copies of that variable and for threads you usually end up with thread safety issues.  The second issue is that events that arrive at exactly the same simulated time can arrive in different orders depending on the partitioning.  In general, when coding components for SST you need to ensure that different event ordering for events arriving at exactly the same time does not effect the result.  We are looking into ways we might be able to guarantee ordering remains that same, but do not have a current solution.  If you adhere to both of these, then generally you'll get the same result for serial and parallel runs.

gvoskuilen, 2020-04-01:
Closing, MacSim is not in elements anymore.

closed 2020-04-01 by gvoskuilen