sstsimulator/sst-elements #78: SST SimpleDistribComponent Has Inconsistent Output on Threaded PDES Core

nmhamster, 2016-01-18:


bjmoor, 2016-01-19:
Can you provide more information and/or a test case?  The test-cases for SimpleDistribComponent each only have a single component in the simulator, so threading doesn't mean much there.

nmhamster, 2016-01-19:
Yes, John is going to add the test outputs to the bugs shortly. We just catalogued them yesterday.

bjmoor, 2016-01-19:
Created a pull request on sst-core.   StopAction was using 'cout', which makes output look like garbage when threads are involved.   Hopefully that will help with this issue.

jpvandy, 2016-01-19:
The diff vs. Reference contains two items of interest:
-----------------
3,5c3
< *** Event queue empty, exiting simulation... ***
< *** Event queue empty, exiting simulation... ***
< Simulation is complete, simulated time: 18.4467 Ms
---
> Simulation is complete, simulated time: 100 ms
-----------------
There are the two extra lines about the "Event queue empty".   Testing could ignore those if that is the correct action.
The reported simulation time seems to reflect something not right.

This information came from test_simpleDistribComponent_expon, and is found in the other distributions as well.

bjmoor, 2016-01-19:
Those notices appear due to using more threads than components in the system.  I would say they are ignore-able.  It might be best to special-case squelching those warnings when nothing was ever put on the thread - I leave that to Scott.

nmhamster, 2016-01-20:
I think we shouldn't allow SST to run with more threads than components, can't we just take the nthread = std::min( requested_threads, # of components ) ?

jpvandy, 2016-01-20:
Is the total unreasonable simulation time a symptom of the same issue or is it something else?

I wouldn't consider it "ignore-able"

bjmoor, 2016-01-20:
No, the time difference is something else, though it might be a related symptom.  Again, @feldergast would be best suited to answer.   I'm guessing 18.4467Ms (mega-seconds?) is probably the "end of time" that shows up when nothing was ever inserted into the other thread's event queues.

jpvandy, 2016-02-24:
There are four tests included in this issue:   Distributions: Expon, Gaussian, Discrete, and the look-up-table.

jwilso, 2016-06-29:
@nmhamster Can you confirm that this is fixed?

jpvandy, 2016-06-29:
Certainly appears to be fixed when I run it today.   No idea when it was fixed.

closed 2016-06-29 by jpvandy