sstsimulator/sst-elements #79: SST Miranda Random Gen Goes to Time Limit in Threaded PDES Test

nmhamster, 2016-01-18:
Goes to time limit, seems to have locked.

bjmoor, 2016-01-19:
Does not lock - just spending all of its time in barriers in ThreadSyncSimpleSkip.  Need to work with @feldergast  to try to reduce barrier calls.

nmhamster, 2016-02-26:
I have changed the tests locally to use a setNoCut() on the L1 cache and re-run the tests. The tests are correctly running but are slow as commented by @bjmoor. I can see significant runtime increase in the randomgen.py test because almost every single operation requires a round trip to memory hence there is a lot of traffic on the links and the long time duration means lots of barriers. Basically this test is running correctly but slow because its a somewhat adversarial case for the multi-threaded core (multithreading shouldn't be used for this type of run). 

John - I think we just need to ensure we don't run this with `-n > 1`. The others seem to perform OK in the testing I am seeing locally.

nmhamster, 2016-02-26:
setNoCut() changes are in PR #134.

nmhamster, 2016-04-12:
Tests pass correctly but take much longer due to frequent sync'ing of the components. This is not a bug but the use of component parallelism which is not recommended.

closed 2016-04-12 by nmhamster