sstsimulator/sst-elements #80: MemSieve has Inconsistent Output with Threaded PDES Core

nmhamster, 2016-01-18:
Wide variation is time output for MemSieve runs when using Threaded PDES Core

jpvandy, 2016-01-20:
Not sure the variation is that wide, but . . .
The simulation time is reported as "5.5 ns", for both n=1 and n=2.
The n=1 table is

sieve, histogram_reads, , Histogram, 5501, 0, 0, 24575, 4096, 6, 16395, 268468325, 2, 2, 2, 0, 0, 1, 0, 0, 0, 1, 0
sieve, histogram_writes, , Histogram, 5501, 0, 0, 24575, 4096, 6, 12267, 83534309, 1, 2, 2, 0, 0, 0, 2, 0, 0, 0, 0

and the n=2 table is 

sieve, histogram_reads, , Histogram, 6000, 0, 0, 24575, 4096, 6, 16395, 268468325, 2, 2, 2, 0, 0, 1, 0, 0, 0, 1, 0
sieve, histogram_writes, , Histogram, 6000, 0, 0, 24575, 4096, 6, 12267, 83534309, 1, 2, 2, 0, 0, 0, 2, 0, 0, 0, 0

The only thing different is the number of pico seconds reported.

bjmoor, 2016-01-20:
This would currently be expected.  You *should* see the same thing if you run with two MPI processes.  It has to do with synchronization times between event queue.   I believe @feldergast had some ideas about how to get the Statistics objects to get "real" finish time, rather than "next sync" finish time.

jwilso, 2016-06-29:
@gvoskuilen  to confirm closure of this issue.

nmhamster, 2016-07-06:
This is expected due to parallel synchronization behavior. Issue closed.

closed 2016-07-06 by nmhamster