sstsimulator/sst-elements #94: Boost Releases > 1.56 do not work with SST.  Need demo case.

jpvandy, 2016-01-29:
Need a test case for Boost serialization failure using version higher than 1.56 !

nmhamster, 2016-02-06:
I think 1.57 still works right? Its working on my Mac but anything else is broken.

jwilso, 2016-06-29:
Currently installing 1.61 Boost on all platforms.  Will send out notice once this is complete so that tests can be run against this version to confirm that this is closed.

nmhamster, 2016-07-05:
@jwilso @jpvandy does it look like the early Boost installs are performing OK?

jwilso, 2016-07-05:
I think there are some 1.61 test cases, but I don’t think it is being evaluated everywhere it is installed.  Nor are we evaluating intermediate revisions between 1.56 and 1.61.

J.

nmhamster, 2016-07-06:
1.56 is used for SST v6.0 release.

closed 2016-07-06 by nmhamster