sstsimulator/sst-elements #107: gcc-5.3.1 yields excessive warning messages on Ubuntu 16.04

jpvandy, 2016-02-08:
The excessive messages are from Boost 1.56 reference to "auto_ptr", which is "deprecated.
This was originally list in issuer #95, which has now been close.

allevin, 2016-06-30:
Ran build tests on Boost-1.56 and Boost-1.61 using both gcc-5.4.0 and gcc-6.1.0.

In both compliers, Boost-1.56 generated the "auto_ptr" warning in both core and elements whereas Boost-1.61 built clean with no warnings.

nmhamster, 2016-07-05:
These are internal to Boost and cannot be fixed by SST. We are planning to move to using /any/ version of Boost with the removal of Serialization.

closed 2016-07-05 by nmhamster

jpvandy, 2016-07-28:
Moved the nightly test to using Boost-1.61.    Same test results as 1.56, but without the many, many warnings.