sstsimulator/sst-elements #108: The scheduler Detailed Network test fails on RedHat-7 VM

jpvandy, 2016-02-08:
The test was running on all other tested platforms.   The failure appears to have to do with over writing the motif-xx.log files.
(These failures can trigger another problem, the SQE timelimit processor really attempts to deal with a time limit for an SST run.   the Detailed network test has a python wrapper that loops thru multiple sst invocations.  Consequently, it can fail to terminate a wrapper loop.)

vjleung, 2016-02-08:
John,

So is it just how the redhat-7 vm handles write permissions by default.

Vitus

nmhamster, 2016-02-08:
I think this is really down to how SST Ember actually does file opening. There is a potential ordering effect here on who owns the final FILE*. Its more of a bug in Ember which is now getting exposed.

nmhamster, 2016-02-09:
John, can you try running this again and see if it restores old behavior at least? There is still an issue where a "job" spans multiple SST ranks.

jpvandy, 2016-02-09:
Nightly seems to have restored the old behavior.

I’m VPN from home at the moment.    Be in after a bit.
Was just trying to run on my sandbox of yesterday, when your email came in

    John

nmhamster, 2016-02-09:
> t-181906422>.
> 08291>.

John,

Can you confirm the test passes on RedHat 7.X?

jpvandy, 2016-02-09:
The nightly FAILED on RedHat-7 as it had previously.   It is looping on an empty motif-3.log file.
After updating ember, I get the same result when I run it on the VM.

The other platforms that run it are happy.
     John

jpvandy, 2016-02-19:
Nothing appears to have changed with respect to the file over write problem.
(The SQE problem of this particular infinite loop has been corrected in the Test Suite.)
Is this a scheduler bug or an ember bug?

vjleung, 2016-02-19:
John,

I have lost track of this a little bit.  The last thing I remember is Si saying the problem is not in the scheduler.

Vitus

nmhamster, 2016-02-19:
This is an issue with Ember. :-(

S.

jpvandy, 2016-04-23:
At this point this is a solid failure on the RedHat-7 VM, COERHEL-7, and and on all three Multi Rank  n=2 nightly test, sst-test, El Capitan, Yosemite.

nmhamster, 2016-04-24:
@jpvandy - can you post the error into the issue please? Thank you!

nmhamster, 2016-04-24:
Ok, no problem, I already have it! Sorry for the spam!

jwilso, 2016-06-29:
@afrodri :  Si should evaluate whether this is still a 6.0.0 issue.

nmhamster, 2016-07-05:
Fixed.

jpvandy, 2016-07-14:
This is not fixed,    it may be deferred, or it may be "not to be fixed", but it's not fixed.

jpvandy, 2016-11-09:
When the openmpi version was updated to 1.8.8, this problem became ubiquitous, across all platforms.  It was fixed, with PR 498 on Nov. 8th.

closed 2016-11-09 by jpvandy