sstsimulator/sst-elements #122: Scheduler tests not run multi thread for n>1

jpvandy, 2016-02-19:
Because the scheduler, including scheduler_DetailedNetwork "is not thread safe".

nmhamster, 2016-02-22:
John - can you describe what's not thread safe in the DetailedNetwork? Is this something we can (easily) fix?

jpvandy, 2016-02-24:
It was possible to allow the Detailed Network test to use multithread only on the ember part of the test, which is where most of the compute time is used and not use multiple threads on the scheduler part of the test.

This is currently regarded as an acceptable state of the Scheduler for Release 6.0.

I am not closing this issue at this time, because there is an SST restriction here that needs to be appropriately documented (somewhere).

nmhamster, 2016-03-14:
Moved to SST v7.0.0 because this will not be completed by release time. Requires more in-depth code changes.

closed 2020-09-30 by allevin