sstsimulator/sst-elements #129: McOpteron test2 current fails on Linux as well as MacOS

jpvandy, 2016-02-22:
Test had not been run in a very long time.  The Suppression of the test on MacOS (Issue #76)  inadvertently omit testing on Linux also.

Test will be explicitly omited on all platforms

nmhamster, 2016-04-05:
@jpvandy can you please post the error output you are seeing here so we can view it? Thank you!

jpvandy, 2016-04-05:
This file was created from a current sandbox.   It is very similar (or identical) with a Jenkins failure on February 21st before we stopped running the test.

[mcO-test2-current-output.txt](https://github.com/sstsimulator/sst-elements/files/205019/mcO-test2-current-output.txt)

nmhamster, 2016-07-05:
Moved to attic, shelving until fixed in the future.

closed 2016-07-05 by nmhamster