sstsimulator/sst-elements #132: "fail to mmap" is causing 6 test failures on Ubuntu 14.04 multithread.

jpvandy, 2016-02-24:
Note this is 14.04 not the bleeding edge 16.04.  The tests are two Mirandas, stencil3d and stream, all three prospero test w/o dramsim and qsim.

(These are in addition to other fails which are also observed on other platforms.)

jpvandy, 2016-03-02:
This appears to not be a SST code issue, but a memory management issue on the Ubuntu 14.04 VM

jwilso, 2016-03-02:
Please the describe the error exactly and provide the value of errno.

jpvandy, 2016-03-05:
The only message from SST for the failures is the one line,
"FATAL: --->  Failed to MMAP backing store for memory: Cannot allocate memory"

This was not a one time occurrence.   It was happening only on the (2cpu) Ubuntu 14.04 VM on the multi-thread n=2 tests.    The first occurrence was the nightly the morning of February 23, The last occurrence was a manually started run on March 1st.   (There was a manual run on February 24 that did not demonstrate the error.)    The root cause is yet unknown.  There was a file cleaning and restart of the VM on March 1st when the problem went away.   Can find no evidence of any individual test consuming quantities of /tmp space.

jpvandy, 2016-03-05:
This Issue should probably be under SQE, instead of elements.

jwilso, 2016-03-07:
I will need errno as reported by the failed mmap call to initiate analysis.  From the mmap manpage:

RETURN VALUE

       On success, mmap() returns a pointer to the mapped area.  On error,  the  value  MAP_FAILED  (that  is,  (void *) -1)  is

       returned,  and  errno is set appropriately.  On success, munmap() returns 0, on failure -1, and errno is set (probably to

       EINVAL).

ERRORS

       EACCES A file descriptor refers to a non-regular file.  Or MAP_PRIVATE was requested, but fd is not open for reading.  Or

              MAP_SHARED  was requested and PROT_WRITE is set, but fd is not open in read/write (O_RDWR) mode.  Or PROT_WRITE is

              set, but the file is append-only.

       EAGAIN The file has been locked, or too much memory has been locked (see setrlimit(2)).

       EBADF  fd is not a valid file descriptor (and MAP_ANONYMOUS was not set).

       EINVAL We don't like addr, length, or offset (e.g., they are too large, or not aligned on a page boundary).

       EINVAL (since Linux 2.6.12) length was 0.

       EINVAL flags contained neither MAP_PRIVATE or MAP_SHARED, or contained both of these values.

       ENFILE The system limit on the total number of open files has been reached.

       ENODEV The underlying filesystem of the specified file does not support memory mapping.

       ENOMEM No memory is available, or the process's maximum number of mappings would have been exceeded.

       EPERM  The prot argument asks for PROT_EXEC but the mapped area belongs to a file on a filesystem that  was  mounted  no-

              exec.

       ETXTBSY

              MAP_DENYWRITE was set but the object specified by fd is open for writing.

       EOVERFLOW

              On 32-bit architecture together with the large file extension (i.e., using 64-bit off_t): the number of pages used

              for length plus number of pages used for offset would overflow unsigned long (32 bits).

       Use of a mapped region can result in these signals:

       SIGSEGV

              Attempted write into a region mapped as read-only.

       SIGBUS Attempted access to a portion of the buffer that does not correspond to the file (for example, beyond the  end  of

              the file, including the case where another process has truncated the file).

jpvandy, 2016-03-07:
The error message has been augmented:

FATAL: --->  Failed to MMAP backing store for memory: Cannot allocate memory, errno = 12

nmhamster, 2016-03-07:
How much memory is being asked for?

jwilso, 2016-03-07:
I think we understand it now.

There are 2 Ubuntu 14.04 VMs.

Ubuntu 14.04 LTS (1)
Ubuntu 14.04 LTS (2)

The Ubuntu 14.04 LTS (1) VM has 1 CPU and 2GB RAM allocated.

The Ubuntu 14.04 LTS (2) VM has 2 CPUs and 4GB RAM allocated.  However, at some point, VMware decided to reset the RAM allocation of this VM back to the default of 2GB.  This is a known problem that we have encountered before where VMware resets the manually specified RAM allocation to its initial recommended value of 2 GB when the VM is stopped and restarted.

I have reallocated 4GB RAM to the Ubuntu 14.04 LTS (2) VM.  John will check to see if this fixes the issue.  If so, I will add VM RAM allocation verification to my checklist for monthly system/security updates, because this would likely be when the memory setting gets reset.

J.

nmhamster, 2016-03-08:
Thanks Jon. Are we going to print some helpful message if we encounter this issue?

jwilso, 2016-03-08:
It is telling us using the errno reported by the failed mmap call.  The test is now printing the value of errno (which was 12 in this case) so that we can know if it fails there for that reason.

I wouldn't expect it to happen on most user systems, as they are not as restricted on resources as our VMs are.  Also, when mmap fails, it announces "mmap failure" pretty loudly; I'm sure that most persons would look at the mmap manpage for further info and trace the errno back just as I did.

Seems to me that we uncovered a test deficiency; we should be evaluating errno and printing it on a system call failure if at all possible to aid in problem analysis.

J.

jwilso, 2016-06-29:
JW: Allocated more VM memory.  Gwen:  Fixed error message as well.

closed 2016-06-29 by jwilso