sstsimulator/sst-elements #135: Miranda and Ariel Examples for Sandy Bridge, Ivy Bridge and KNC are Not in Distribution

nmhamster, 2016-02-26:
Need to provide example Python files for our SNB, IVB and KNC models.

jpvandy, 2016-03-02:
The example inputs for Ariel are there.  There simply buried more deeply in  

 elements/ariel/frontend/simple/examples/stream/

jwilso, 2016-06-29:
Arun: Should make these easier to find, and add KNC.

nmhamster, 2016-07-12:
Decision made to post these to the website *after* the SST v6.0 release has been posted and then to work on including them in future releases. Changed to SST v6.1.0.

nmhamster, 2016-12-19:
These will be posted in an online repository of models to be added during SST v7.0, not being done for SST v6.0/v6.1.

hughes-c, 2019-07-17:
@nmhamster Ivy Bridge and Sandy Bridge models are in the ariel test folder and have been there since https://github.com/sstsimulator/sst-elements/commit/87130be2c4bd6e3ae5120cf9d847912f7c39a07f. We have a KNL model that we can add.

allevin, 2020-09-30:
@gvoskuilen  to take care of this.

hughes-c, 2023-10-24:
Closing this. Will update with more modern processor models.

closed 2023-10-24 by hughes-c