sstsimulator/sst-elements #147: Ember Sweep has non-reproducible one time failures

jpvandy, 2016-03-08:
Here the one from multithread=2, sst-test CentOS on March 8th:    (This is the first I have noticed from Firefly.)

test_EmberSweep_ce2d511d03dc1083254fdb8149d5d78d
 151 run, 0 have failed
      dragonfly2 --shape=8:8:4:8  AllPingPong iterations=10 messageSize=20000 
~/jenkins/workspace/SST__x-MultiThread_sst-test-n2/155_2016-03-08_04-17-00/devel/trunk/sst/elements/ember/test ~/jenkins/workspace/SST__x-MultiThread_sst-test-n2/155_2016-03-08_04-17-00/devel/trunk
sst: merlinEvent.h:52: void SST::Firefly::FireflyNetworkEvent::bufPop(size_t): Assertion `offset <= bufLen' failed.
[sst-test:05163] *** Process received signal ***
[sst-test:05163] Signal: Aborted (6)
[sst-test:05163] Signal code:  (-6)
[sst-test:05163] [ 0] /lib64/libpthread.so.0(+0xf100)[0x7f2ed0405100]
[sst-test:05163] [ 1] /lib64/libc.so.6(gsignal+0x37)[0x7f2ece29f5f7]
[sst-test:05163] [ 2] /lib64/libc.so.6(abort+0x148)[0x7f2ece2a0ce8]
[sst-test:05163] [ 3] /lib64/libc.so.6(+0x2e566)[0x7f2ece298566]
[sst-test:05163] [ 4] /lib64/libc.so.6(+0x2e612)[0x7f2ece298612]
[sst-test:05163] [ 5] /home/jwilso/jenkins/workspace/SST__x-MultiThread_sst-test-n2/155_2016-03-08_04-17-00/local/lib/sst/libfirefly.so(+0x488b0)[0x7f2ebbc1d8b0]
[sst-test:05163] [ 6] /home/jwilso/jenkins/workspace/SST__x-MultiThread_sst-test-n2/155_2016-03-08_04-17-00/local/lib/sst/libfirefly.so(_ZN3SST7Firefly3Nic11RecvMachine7state_1EPNS0_19FireflyNetworkEventE+0x1d0c)[0x7f2ebbc8411c]
[sst-test:05163] [ 7] /home/jwilso/jenkins/workspace/SST__x-MultiThread_sst-test-n2/155_2016-03-08_04-17-00/local/lib/sst/libfirefly.so(_ZN3SST7Firefly3Nic15handleSelfEventEPNS_5EventE+0x1b)[0x7f2ebbc715fb]
[sst-test:05163] [ 8] sst(_ZN3SST10Simulation3runEv+0x192)[0x68d852]
[sst-test:05163] [ 9] sst[0x5c9443]
[sst-test:05163] [10] /lib64/libstdc++.so.6(+0xb5220)[0x7f2ecebf8220]
[sst-test:05163] [11] /lib64/libpthread.so.0(+0x7dc5)[0x7f2ed03fddc5]
[sst-test:05163] [12] /lib64/libc.so.6(clone+0x6d)[0x7f2ece36021d]
[sst-test:05163] *** End of error message ***
/home/jwilso/jenkins/workspace/SST__x-MultiThread_sst-test-n2/155_2016-03-08_04-17-00/devel/trunk/sst/elements/ember/test/bashIN: line 901:  5163 Aborted                 (core dumped) sst -n 2 --model-options="--topo=dragonfly2 --shape=8:8:4:8  --cmdLine=\"Init\" --cmdLine=\"AllPingPong iterations=10 messageSize=20000 \" --cmdLine=\"Fini\"" emberLoad.py > tmp_file
       SST run is incomplete, FATAL
shunit.ASSERT:  # 151: SST run is incomplete, FATAL
 
 
 12  37 371 /home/jwilso/jenkins/workspace/SST__x-MultiThread_sst-test-n2/155_2016-03-08_04-17-00/devel/trunk/sst/elements/ember/test/tmp_file
    ----   stdout for sst:
EMBER: platform: default
8
8
4
8
512
EMBER: network: topology=dragonfly2 shape=8:8:4:8
EMBER: numNodes=512 numNics=512
EMBER: network: BW=4GB/s pktSize=2048B flitSize=8B
EMBER: Job: -nidList=0-511 -ranksPerNode=1 ['Init']
EMBER: Job: -nidList=0-511 -ranksPerNode=1 ['AllPingPong', 'iterations=10', 'messageSize=20000']
EMBER: Job: -nidList=0-511 -ranksPerNode=1 ['Fini']
    ----   end of stdout 
 
151: Wall Clock Time  12 sec.   dragonfly2 --shape=8:8:4:8  AllPingPong iterations=10 messageSize=20000 
 
~/jenkins/workspace/SST__x-MultiThread_sst-test-n2/155_2016-03-08_04-17-00/devel/trunk
         -------

jpvandy, 2016-03-08:
Here's one from March 5th, same multithread=2, sst-test CentOS:
Test #1 of 180 in the sweep.
(Same Assertion about BufLen  from merlinEvent.h:52)

test_EmberSweep_f9148701b5882168aef0a43d84f06535
 1 run, 0 have failed
      torus --shape=16x16x16  AllPingPong iterations=10 messageSize=20000 
~/jenkins/workspace/SST__x-MultiThread_sst-test-n2/152_2016-03-05_04-17-00/devel/trunk/sst/elements/ember/test ~/jenkins/workspace/SST__x-MultiThread_sst-test-n2/152_2016-03-05_04-17-00/devel/trunk
sst: merlinEvent.h:52: void SST::Firefly::FireflyNetworkEvent::bufPop(size_t): Assertion `offset <= bufLen' failed.
[sst-test:05069] *** Process received signal ***
[sst-test:05069] Signal: Aborted (6)
[sst-test:05069] Signal code:  (-6)
[sst-test:05069] [ 0] /lib64/libpthread.so.0(+0xf100)[0x7f876db14100]
[sst-test:05069] [ 1] /lib64/libc.so.6(gsignal+0x37)[0x7f876b9ae5f7]
[sst-test:05069] [ 2] /lib64/libc.so.6(abort+0x148)[0x7f876b9afce8]
[sst-test:05069] [ 3] /lib64/libc.so.6(+0x2e566)[0x7f876b9a7566]
[sst-test:05069] [ 4] /lib64/libc.so.6(+0x2e612)[0x7f876b9a7612]
[sst-test:05069] [ 5] /home/jwilso/jenkins/workspace/SST__x-MultiThread_sst-test-n2/152_2016-03-05_04-17-00/local/lib/sst/libfirefly.so(+0x488b0)[0x7f87592a78b0]
[sst-test:05069] [ 6] /home/jwilso/jenkins/workspace/SST__x-MultiThread_sst-test-n2/152_2016-03-05_04-17-00/local/lib/sst/libfirefly.so(_ZN3SST7Firefly3Nic11RecvMachine7state_1EPNS0_19FireflyNetworkEventE+0x1d0c)[0x7f875930e11c]
[sst-test:05069] [ 7] /home/jwilso/jenkins/workspace/SST__x-MultiThread_sst-test-n2/152_2016-03-05_04-17-00/local/lib/sst/libfirefly.so(_ZN3SST7Firefly3Nic15handleSelfEventEPNS_5EventE+0x1b)[0x7f87592fb5fb]
[sst-test:05069] [ 8] sst(_ZN3SST10Simulation3runEv+0x192)[0x68d852]
[sst-test:05069] [ 9] sst[0x5c9443]
[sst-test:05069] [10] /lib64/libstdc++.so.6(+0xb5220)[0x7f876c307220]
[sst-test:05069] [11] /lib64/libpthread.so.0(+0x7dc5)[0x7f876db0cdc5]
[sst-test:05069] [12] /lib64/libc.so.6(clone+0x6d)[0x7f876ba6f21d]
[sst-test:05069] *** End of error message ***
/home/jwilso/jenkins/workspace/SST__x-MultiThread_sst-test-n2/152_2016-03-05_04-17-00/devel/trunk/sst/elements/ember/test/bashIN: line 1:  5069 Aborted                 (core dumped) sst -n 2 --model-options="--topo=torus --shape=16x16x16  --cmdLine=\"Init\" --cmdLine=\"AllPingPong iterations=10 messageSize=20000 \" --cmdLine=\"Fini\"" emberLoad.py > tmp_file
       SST run is incomplete, FATAL
shunit.ASSERT:  # 1: SST run is incomplete, FATAL
 
 
  7  32 360 /home/jwilso/jenkins/workspace/SST__x-MultiThread_sst-test-n2/152_2016-03-05_04-17-00/devel/trunk/sst/elements/ember/test/tmp_file

Another similar on the test #1on March 6 on Ubuntu 14.04 n=2

And on Ubuntu on March 3 test #147 went time limit.

Finally on the RHEL-6 VM n=2, on February 23 #1 went time limit.

jpvandy, 2016-04-05:
I put some instrumentation in here's where and the failing values

    void bufPop( size_t len ) {

        offset += len;
        if ( offset > bufLen ) {
        std::cerr << len << " " << offset << " " << bufLen << std::endl;
        assert( offset <= bufLen );
        }
    }
   
 firefly/merlinEvent.h    near line 50

Instrumentation is mine, NOT in devel.

When it fails, I see   36  36  24
len=36  offset=36   bufLen =24

That's all I know.

jpvandy, 2016-04-22:
Most recent failure of this type was April 16th on Test #1 of the 180 tests in EmberSweep

(There was a single test time-limit today on test #44.    Not the same.)

jpvandy, 2016-04-25:
There also was a bufLen failure on ES test #9 on April 22, on the Ubuntu VM, running nThreads=2.

jpvandy, 2016-04-25:
And a bufLen failure on ES test #151 on April 21, on the Ubuntu VM, nthreads=2

jpvandy, 2016-05-12:
And a bufLen failure on ES test #1 on May 12 on the Red Hat 6 VM for nthreads =2

jpvandy, 2016-05-26:
BufLen failure on ES test #1 on sst-test (single thread) on May 26th

jpvandy, 2016-06-02:
BufLen failure on EmberSweep test #1 on June 2nd on sst-test 
on Ariel tester run #1862. it is a nthread = 2 run.

jpvandy, 2016-06-08:
One occurrence in the Nightly test #1  June 8th on Ariel tester #1912 on sst-test.  It is a nthread = 2 run.

jpvandy, 2016-06-13:
Another this AM, June 13th, on Ubuntu 14.04 Ariel tester (570), nthread=2

jpvandy, 2016-06-15:
Midnight last night  June 14/15, another nthread=2 on sst-test  on Ariel test #1

jpvandy, 2016-06-23:
June 23, on MacOS multi-thread=2  on EmberSweep test 151

jpvandy, 2016-06-23:
June 23, on sst-test 1220 hours on "Ariel tester"  EmberSweep test #1

jpvandy, 2016-06-29:
June 29, on Ariel tester, that started at 3:12 AM  EmberSweep test #1 had a bufLen assert.  (this is nthread=2)

jpvandy, 2016-07-06:
July 6th nightly, on HREL-6, multi-thread = 2, EmberSweep test #9 had a bufLen assert.

jpvandy, 2016-07-12:
July 11th, on sst-test (Ariel tester), EmberSweep test #1 had a bufLen assert,  The is multi-thread = 2 and 127 seconds of Wall Clock time into the sst execution

jpvandy, 2016-08-01:
Skip in recording these.     August 1st, multiThread=2  test #1 caught one  162 seconds of Wall clock in.

jpvandy, 2016-08-05:
August 4th and 5th, multiThread=2, on sst-test_Ariel_tester, test #1 on runs 2367, 2368 and 2371.

jpvandy, 2016-08-11:
August 11th, multiThread=2, on sst-test_Ariel_tester, test #1 on run 2418.   (This is after the ember/firefly changes to make Valgrind happy.)   Wall clock is 144 seconds.

jpvandy, 2016-08-15:
August 15th, test #1 failed on the "no-mpi" test project  (single thread).

jpvandy, 2016-08-15:
Two failures yesterday morning, August 14th, on instrumented runs.     See above April 5th for instrumentation.
The second one from the run of Ariel-tester that started at 6:12 had the relatively uninteresting debug output of "36 36 0" 
The first, started at 3:12 had very different debug output:
36 8589934631 446760672

jpvandy, 2016-08-16:
August 16,  four failures overnight on sst-test.  Three had the normal debug out put, I believe "36 39 36", but one had a new different debug output, "3652 3688 0"

jpvandy, 2016-08-17:
36 8589934631 468560064 another different unique output from debug instrumentation occurring once in overnight on August 16-17.   There were a total of 5 failures over night.  All on EmberSweep test #1

jpvandy, 2016-08-18:
August 18th, an uninstrumented bufLen failure on El Capitan nthread=2,  This was on EmberSweep test #151, (which is one of the three test where the error has been observed in sst-test).

There were also three overnight on the instrumented run on sst-test.  They are nthread=2 and these occurred on test #1.

jpvandy, 2016-10-04:
Stopped logging nightly failures.

Have a Jenkins Project (with nthread=2) that just runs multiple passes of EmberSweep tests 1, 9 and 151 and gets a more than 25%  failure rate on the passes.      This test is done with a small amount of Ember element instrumentation and in this case adding instrumentation significantly boosted the failure rate.

jpvandy, 2016-11-11:
Is this really an Ember problem or somebody else's?
@nmhamster 
@feldergast 
@mjleven 
test_EmberSweep_1_f9148701b5882168aef0a43d84f06535
 1 run, 0 have failed
      torus --shape=16x16x16  AllPingPong iterations=10 messageSize=20000 
~/jenkins/workspace/SST__Nightly_MultiThread_sst-test-n2/454_454/devel/trunk/sst-elements/src/sst/elements/ember/test ~/jenkins/workspace/SST__Nightly_MultiThread_sst-test-n2/454_454/devel/trunk
sst: merlinEvent.h:56: void SST::Firefly::FireflyNetworkEvent::bufPop(size_t): Assertion `offset <= bufLen' failed.
[sst-test:29264] *** Process received signal ***
[sst-test:29264] Signal: Aborted (6)
[sst-test:29264] Signal code:  (-6)
[sst-test:29264] [ 0] /lib64/libpthread.so.0(+0xf100)[0x7fa627ffa100]
[sst-test:29264] [ 1] /lib64/libc.so.6(gsignal+0x37)[0x7fa625e915f7]
[sst-test:29264] [ 2] /lib64/libc.so.6(abort+0x148)[0x7fa625e92ce8]
[sst-test:29264] [ 3] /lib64/libc.so.6(+0x2e566)[0x7fa625e8a566]
[sst-test:29264] [ 4] /lib64/libc.so.6(+0x2e612)[0x7fa625e8a612]
[sst-test:29264] [ 5] /home/jwilso/jenkins/workspace/SST__Nightly_MultiThread_sst-test-n2/454_454/local/sst-elements/lib/sst-elements-library/libfirefly.so(+0x91173)[0x7fa613a0c173]
[sst-test:29264] [ 6] /home/jwilso/jenkins/workspace/SST__Nightly_MultiThread_sst-test-n2/454_454/local/sst-elements/lib/sst-elements-library/libfirefly.so(_ZN3SST7Firefly3Nic11RecvMachine7state_1EPNS0_19FireflyNetworkEventE+0x65a)[0x7fa613a0dcda]
[sst-test:29264] [ 7] /home/jwilso/jenkins/workspace/SST__Nightly_MultiThread_sst-test-n2/454_454/local/sst-elements/lib/sst-elements-library/libfirefly.so(_ZN3SST7Firefly3Nic15handleSelfEventEPNS_5EventE+0x1b)[0x7fa613a0197b]
[sst-test:29264] [ 8] sst(_ZN3SST10Simulation3runEv+0x1bf)[0x4b58ef]
[sst-test:29264] [ 9] sst[0x466d88]
[sst-test:29264] [10] /lib64/libstdc++.so.6(+0xb5220)[0x7fa6267eb220]
[sst-test:29264] [11] /lib64/libpthread.so.0(+0x7dc5)[0x7fa627ff2dc5]
[sst-test:29264] [12] /lib64/libc.so.6(clone+0x6d)[0x7fa625f52ced]
[sst-test:29264] *** End of error message ***
/home/jwilso/jenkins/workspace/SST__Nightly_MultiThread_sst-test-n2/454_454/devel/trunk/sst-elements/src/sst/elements/ember/test/bashIN: line 1: 29264 Aborted                 (core dumped) sst -n 2 --model-options="--topo=torus --shape=16x16x16  --cmdLine=\"Init\" --cmdLine=\"AllPingPong iterations=10 messageSize=20000 \" --cmdLine=\"Fini\"" emberLoad.py > tmp_file
       SST run is incomplete, FATAL
shunit.ASSERT:  # 1: SST run is incomplete, FATAL
 
 
  7  32 360 /home/jwilso/jenkins/workspace/SST__Nightly_MultiThread_sst-test-n2/454_454/devel/trunk/sst-elements/src/sst/elements/ember/test/tmp_file
    ----   stdout for sst:
EMBER: platform: default
EMBER: network: topology=torus shape=16x16x16
EMBER: numNodes=4096 numNics=4096
EMBER: network: BW=4GB/s pktSize=2048B flitSize=8B
EMBER: Job: -nidList=0-4095 -ranksPerNode=1 ['Init']
EMBER: Job: -nidList=0-4095 -ranksPerNode=1 ['AllPingPong', 'iterations=10', 'messageSize=20000']
EMBER: Job: -nidList=0-4095 -ranksPerNode=1 ['Fini']
    ----   end of stdout 
 
1: Wall Clock Time  156 sec.   torus --shape=16x16x16  AllPingPong iterations=10 messageSize=20000

jpvandy, 2016-11-11:
@nmhamster 
@mjleven 
@feldergast 
This is a low probability, Nthreads> 1 problem.  It fails probably less than once a week.

However I have a elements branch that ups the probability to more like 30%.

nmhamster, 2016-12-19:
Marked as a known issue for SST v6.1.

jpvandy, 2017-02-22:
This appears to have been fixed by Mike with PR #596.
@nmhamster
@allevin 
@mjleven

closed 2017-02-22 by jpvandy