sstsimulator/sst-elements #151: SST Ariel/PIN Tests are only on sst-test

nmhamster, 2016-03-14:
We are only testing Ariel/PIN on `sst-test`. This needs to be expanded to include:

- Ubuntu
- Mac OSX

@jpvandy says that the PIN download is only installed on `sst-test`.

jwilso, 2016-03-14:
Is PIN available for Xcode 7.x and GCC versions 4.8.4 and 5.3.1 (Ubuntu 14.04 and 16.04)?

Thanks,

J.

nmhamster, 2016-03-14:
@jwilso - good question. The current "Linux" version should work correctly with GCC 4.8.4 (I am running that on my workstation and its fine). For GCC 5.3.1 I'm not sure that this is going to work but I can test it out. On Mac OSX the standard download should also work with XCode 7.X, I'm running this on my laptop without any issues.

jwilso, 2016-03-14:
I'll work on installing PIN r71313 to Ubuntu and Mac OS X per the downloads listed here.  I'm skeptical as to whether it'll work on GCC 5.x, but I suppose we'll find out.

https://software.intel.com/en-us/articles/pin-a-binary-instrumentation-tool-downloads

J.

jpvandy, 2016-03-15:
The pin library is now there on Ubuntu, but the Ariel run fails.   All six tests fail the same
(There were three Jenkins projects that tried. Two on 1-cpu, and one on 2-cpu.)

test_Ariel_runstream
 starting Directory /home/sstbuild/jenkins/workspace/SST__git_on_VM-make_dist_Ubuntu_14.04/161_2016-03-15_02-43-00/devel/trunk/distTestDir/trunk
/home/sstbuild/jenkins/workspace/SST__git_on_VM-make_dist_Ubuntu_14.04/161_2016-03-15_02-43-00/devel/trunk/distTestDir/trunk/sst/elements/ariel/frontend/simple/examples/stream/runstream.py
FATAL: ArielComponent[arielcpu.cc:395:forkPINChild] Launching trace child failed!  Exited with status 0
 
WARNING: sst did not finish normally, RetVal= 1
 
shunit.ASSERT: WARNING: sst did not finish normally, RetVal=1

jwilso, 2016-03-15:
Exit status zero indicates success, not failure.  Actual generated errno at point of failure needs to be propagated or mapped against errno.h and displayed.

J.

https://github.com/sstsimulator/sst-elements/issues/151#issuecomment-196918039

jpvandy, 2016-03-18:
Pin files are install on Yosemite at this point. The build of stream is failing over not finding gomp.  This seems to be an openMP library and the Web says the openMP is supported on Clang.

nmhamster, 2016-03-31:
I think we may want to run without OpenMP on Mac OS X until we get the support merged in (it has recently been added in upstream LLVM).

nmhamster, 2016-03-31:
@jpvandy You may also want to look at PR #201 for the PIN fixes on Mac OS X for Prospero.

nmhamster, 2016-07-05:
Gradually being rolled out across testing platforms.

closed 2016-07-05 by nmhamster