sstsimulator/sst-elements #162: 9 tests fail when "mpirun -np 2" is applied to entire collection of 200+ tests

jpvandy, 2016-03-16:
This is on sst-test.   It is believed that all tests Suites are considered, (except zoltan and partitioner, which already use mpirun and Sched_DetailedNetwork which which is a special case with mixed threading and not.)

1. test Sirius_Zodiac_27 fails wiith a C++ throw.  (see traceback below)
2. memHSiieve fails but with no output or error message.
3. Simple component (Issue #148) fails with 19 lines in stead of the expected 9.
4. miranda singlestream fails with a 200 second time Limit (Not yet retried with much large time.
5. Miranda randomgen fail with a 3500 second time limit.  (Not repeated yet with longer time.)
6. simple statistics fails with same error as with Multithread n=2, (Issue ??)
7-8. VaultSim tests fail with unregistered component.  (See traceback below)
9. Qsim fails with time limt. (perhaps not really an mpirun issue.)

1. test Sirius_Zodiac_27 fails wiith a C++ throw.;
test_Sirius_Zodiac_27
~/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/devel/trunk/sst/elements/zodiac/test/allreduce ~/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/devel/trunk
Need sutArgs
--model-options "--shape=27" /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/devel/trunk/sst/elements/zodiac/test/allreduce/allreduce.py
------------------------------
 
WARNING: sst did not finish normally
 
-rwxr-xr-x 1 jwilso jwilso 29463 Mar 16 04:30 /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst
  491  1722 25625 /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/devel/trunk/test/testOutputs/test_Sirius_allred_27.tmp
   54   432  2896 /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/devel/trunk/test/testReferenceFiles/test_Sirius_allred_27.out
  545  2154 28521 total
shunit.ASSERT:  WARNING: sst did not finish normally, RetVal=134
terminate called after throwing an instance of 'boost::archive::archive_exception'
  what():  unregistered class - derived class not registered or exported
[sst-test:22911] *** Process received signal ***
[sst-test:22911] Signal: Aborted (6)
[sst-test:22911] Signal code:  (-6)
[sst-test:22911] [ 0] /lib64/libpthread.so.0(+0xf100)[0x7f054c0a5100]
[sst-test:22911] [ 1] /lib64/libc.so.6(gsignal+0x37)[0x7f0549f3f5f7]
[sst-test:22911] [ 2] /lib64/libc.so.6(abort+0x148)[0x7f0549f40ce8]
[sst-test:22911] [ 3] /lib64/libstdc++.so.6(_ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x165)[0x7f054a8439d5]
[sst-test:22911] [ 4] /lib64/libstdc++.so.6(+0x5e946)[0x7f054a841946]
[sst-test:22911] [ 5] /lib64/libstdc++.so.6(+0x5e973)[0x7f054a841973]
[sst-test:22911] [ 6] /lib64/libstdc++.so.6(+0x5eb93)[0x7f054a841b93]
[sst-test:22911] [ 7] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst[0x5e5869]
[sst-test:22911] [ 8] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(_ZNK5boost7archive6detail11oserializerINS0_20polymorphic_oarchiveESt6vectorIPN3SST8ActivityESaIS7_EEE16save_object_dataERNS1_14basic_oarchiveEPKv+0x320)[0x6ad4c0]
[sst-test:22911] [ 9] /usr/local/module-pkgs/boost/boost-1.56.0-nompi_gcc-4.8.3/lib/libboost_serialization.so.1.56.0(_ZN5boost7archive6detail14basic_oarchive11save_objectEPKvRKNS1_17basic_oserializerE+0x19d)[0x7f054b69affd]
[sst-test:22911] [10] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(_ZN3SST9SyncQueue7getDataEv+0x3d9)[0x6ac749]
[sst-test:22911] [11] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(_ZN3SST18RankSyncSerialSkip8exchangeEv+0x121)[0x6875c1]
[sst-test:22911] [12] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(_ZN3SST18RankSyncSerialSkip7executeEi+0xc5)[0x688015]
[sst-test:22911] [13] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(_ZN3SST11SyncManager7executeEv+0x6b)[0x6abc5b]
[sst-test:22911] [14] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(_ZN3SST10Simulation3runEv+0x192)[0x68d852]
[sst-test:22911] [15] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst[0x5c9443]
[sst-test:22911] [16] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(main+0x1cc6)[0x5bcb56]
[sst-test:22911] [17] /lib64/libc.so.6(__libc_start_main+0xf5)[0x7f0549f2bb15]
[sst-test:22911] [18] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst[0x5c8491]
[sst-test:22911] *** End of error message ***
         -------
2. memHSiieve fails but with no output or error message.
3. Simple component (Issue #148) fails with 19 lines in stead of the expected 9.
4. miranda singlestream fails with a 200 second time Limit (Not yet retried with much large time.
5. Miranda randomgen fail with a 3500 second time limit.  (Not repeated yet with longer time.)
6. simple statistics fails with same error as with Multithread n=2, (Issue ??)
7. VaultSim tests fail with unregistered component.
test_VaultSim_sdl2
terminate called after throwing an instance of 'boost::archive::archive_exception'
  what():  unregistered class - derived class not registered or exported
[sst-test:10960] *** Process received signal ***
[sst-test:10960] Signal: Aborted (6)
[sst-test:10960] Signal code:  (-6)
[sst-test:10960] [ 0] /lib64/libpthread.so.0(+0xf100)[0x7f0269c3c100]
[sst-test:10960] [ 1] /lib64/libc.so.6(gsignal+0x37)[0x7f0267ad65f7]
[sst-test:10960] [ 2] /lib64/libc.so.6(abort+0x148)[0x7f0267ad7ce8]
[sst-test:10960] [ 3] /lib64/libstdc++.so.6(_ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x165)[0x7f02683da9d5]
[sst-test:10960] [ 4] /lib64/libstdc++.so.6(+0x5e946)[0x7f02683d8946]
[sst-test:10960] [ 5] /lib64/libstdc++.so.6(+0x5e973)[0x7f02683d8973]
[sst-test:10960] [ 6] /lib64/libstdc++.so.6(+0x5eb93)[0x7f02683d8b93]
[sst-test:10960] [ 7] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst[0x5e5869]
[sst-test:10960] [ 8] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(_ZNK5boost7archive6detail11oserializerINS0_20polymorphic_oarchiveESt6vectorIPN3SST8ActivityESaIS7_EEE16save_object_dataERNS1_14basic_oarchiveEPKv+0x320)[0x6ad4c0]
[sst-test:10960] [ 9] /usr/local/module-pkgs/boost/boost-1.56.0-nompi_gcc-4.8.3/lib/libboost_serialization.so.1.56.0(_ZN5boost7archive6detail14basic_oarchive11save_objectEPKvRKNS1_17basic_oserializerE+0x19d)[0x7f0269231ffd]
[sst-test:10960] [10] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(_ZN3SST9SyncQueue7getDataEv+0x3d9)[0x6ac749]
[sst-test:10960] [11] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(_ZN3SST18RankSyncSerialSkip8exchangeEv+0x121)[0x6875c1]
[sst-test:10960] [12] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(_ZN3SST18RankSyncSerialSkip7executeEi+0xc5)[0x688015]
[sst-test:10960] [13] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(_ZN3SST11SyncManager7executeEv+0x6b)[0x6abc5b]
[sst-test:10960] [14] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(_ZN3SST10Simulation3runEv+0x192)[0x68d852]
[sst-test:10960] [15] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst[0x5c9443]
[sst-test:10960] [16] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst(main+0x1cc6)[0x5bcb56]
[sst-test:10960] [17] /lib64/libc.so.6(__libc_start_main+0xf5)[0x7f0267ac2b15]
[sst-test:10960] [18] /home/jwilso/jenkins/workspace/SST__zzz_USEMPIRUN_sst-test_full/6_2016-03-16_04-17-00/local/bin/sst[0x5c8491]
[sst-test:10960] *** End of error message ***
 
WARNING: sst did not finish normally

9. Qsim fails with time limlt. (perhaps not really an mpirun issue.)

feldergast, 2016-03-16:
Initial triage:

+ test Sirius_Zodiac_27 fails wiith a C++ throw. (see traceback below)

- Uses an event that does not implement serialization.  No runnable in parallel for this reason

+ memHSiieve fails but with no output or error message.

- This one needs more study, but guessing it has something to do with PIN interactions.

+ Simple component (Issue #148<https://github.com/sstsimulator/sst-elements/issues/148>) fails with 19 lines in stead of the expected 9.

- Need to look into this more

+ miranda singlestream fails with a 200 second time Limit (Not yet retried with much large time.

- Shouldn't be run in parallel (too much communication, not enough processing to amortize synchronization costs)

+ Miranda randomgen fail with a 3500 second time limit. (Not repeated yet with longer time.)

- Shouldn't be run in parallel (too much communication, not enough processing to amortize synchronization costs)

+ simple statistics fails with same error as with Multithread n=2, (Issue ??) 7-8. VaultSim tests fail with unregistered component. (See traceback below)

-  Need to look into this one more

+ Qsim fails with time limt. (perhaps not really an mpirun issue.)

- Not sure if Qsim should be run in parallel or not.  Probably want to check with developer

gvoskuilen, 2016-03-16:
The MemHSieve fail may be related to the btmalloc library preloading since that's caused similar failures elsewhere. We're working on an alternative to avoid btmalloc so I'd not worry too much about this fail until the new code/test is in. Probably later this week.

nmhamster, 2016-04-12:
@jpvandy can we get the latest count on failing tests, I think now almost all of these are passing correctly for MPI?

jpvandy, 2016-04-29:
The only test that is failing every time is the schedulerDetailedNetworktest.  This failure is an Ember failure, see issue #108.   The listed 9 have been resolved.
The Qsim test is not being run multi core. A couple tests out of large Suites are being skipped as having features that don't serialize.   MemHSieve is getting a "soft" pass, but a signficant upgrade is anticipated.

closed 2016-04-29 by jpvandy