sstsimulator/sst-elements #228: When run with both 2 threads and 2 ranks, all ember test go time-limit

jpvandy, 2016-04-13:
This is the dominant testing failure with multiple threads and ranks

nmhamster, 2016-04-13:
@feldergast are we expected for the N=2/N=2 case to be working yet?

jpvandy, 2016-04-13:
2 x 2 currently has so many ember caused failures, that it's difficult to look for other issues in the test results.    Ember's impact is NOT limited to ember nightly and ember Sweep.

There may be a clue in debugging the ember problem in that the simplest test in embernightly, which was supposed to only look at parameter processing, also goes time limit

jpvandy, 2017-05-01:
The round of core changes last month appear to have fixed this.
No EmberSweep failures noted in the weekly 2x2 Sweep

closed 2017-05-01 by jpvandy