sstsimulator/sst-elements #248: Descripency using prefetcher between single thread and multi thread

jpvandy, 2016-04-20:
Several existing test that use the prefetcher show slightly different result for single thread and thread greater than one.

(Gwen's comment: It's an error somewhere in the way latency is recorded and just manifests when prefetching is on in these cases, not the simulation doing something different (event ordering and timestamps are all identical). I'm tracking it down. Also, it looks like the n=1 case is the wrong one!)

A snippet of test output:
test_miranda_copybench
  213  2763 19216  testReferenceFiles/test_miranda_copybench.out
  213  2763 19216  testOutputs/test_miranda_copybench.out
 
  8  60 446 diff_sorted
Output does not match Reference File
shunit.ASSERT: Output does not match Reference File
186c186
<  l1cache.latency_GetS_IS : Accumulator : Sum.u64 = 31491943; SumSQ.u64 = 76211954019; Count.u64 = 14654; 
---
>  l1cache.latency_GetS_IS : Accumulator : Sum.u64 = 29392821; SumSQ.u64 = 58971798079; Count.u64 = 14654; 
188c188
<  l1cache.latency_GetX_IM : Accumulator : Sum.u64 = 3470545; SumSQ.u64 = 6962249649; Count.u64 = 1730; 
---
>  l1cache.latency_GetX_IM : Accumulator : Sum.u64 = 3470209; SumSQ.u64 = 6960897585; Count.u64 = 1730; 
Simulation is complete, simulated time: 7.31896 ms

jpvandy, 2016-04-20:
This discrepancy is applicable to both multiThread and multiRank.

gvoskuilen, 2016-04-25:
@jpvandy is this resolved now with PR #252?

jpvandy, 2016-04-25:
@gvoskuilen It is Resolved.

closed 2016-04-25 by jpvandy