sstsimulator/sst-elements #269: Motif: Halo3D26 Seg Fault in Firefly

h4u5, 2016-04-29:
I'm running a large Halo3D26 and getting running out of allocated time and getting SegFaults.
This may not be a bug (if I'm just not giving the simulation a long enough time to finish), but I figure I'd open it as an issue, while I continue to explore.

I was allocating 3 hours on 512 sandybridge cores, I'll up this to 10 hours and see if it works:
EMBER: network: topology=dragonfly2 shape=24:48:12:96 arbitration=xbar_arb_lru, routing=adaptive-local
EMBER: numNodes=110592 numNics=110592
EMBER: network: BW=12.5GB/s pktSize=2048B flitSize=32B
EMBER: Job: -nidList=0-110591 -ranksPerNode=1 ['Init']
EMBER: Job: -nidList=0-110591 -ranksPerNode=1 ['Halo3D26', 'nx=100', 'ny=100', 'nz=100', 'pex=48', 'pey=48', 'pez=48', 'peflops=5000000000000', 'flopspercell=34']
EMBER: Job: -nidList=0-110591 -ranksPerNode=1 ['Fini']
Round robin partitioning
Halo3D processor decomposition solution: 48x48x48
Halo3D problem size: 100x100x100
Halo3D compute time: 0 ns
Halo3D copy time:    0 ns
Halo3D iterations:   1
Halo3D items/cell:   1
Halo3D do reduction: 1

[sb1066:150005] *** Process received signal ***
[sb1066:150005] Signal: Segmentation fault (11)
[sb1066:150005] Signal code:  (18168936)
[sb1066:150005] Failing at address: 0x2a268270
[sb1066:150005] [ 0] /lib64/libc.so.6(+0x326a0)[0x2aaaaca416a0]
[sb1066:150005] [ 1] /ascldap/users/tgroves/sst/lib/sst/libfirefly.so(_ZN3SST7Firefly7CtrlMsg18ProcessQueuesStateINS1_3XXXEE16processWaitCtx_1EPNS4_7WaitCtxEPNS1_8_CommReqE+0x146)[0x2aaac5266a16]
[sb1066:150005] [ 2] /ascldap/users/tgroves/sst/lib/sst/libfirefly.so(_ZN3SST7Firefly7CtrlMsg3XXX12delayHandlerEPNS_5EventE+0x47)[0x2aaac5251ab7]
[sb1066:150005] [ 3] sst(_ZN3SST4Link12deliverEventEPNS_5EventE+0x32)[0x90b9a4]
[sb1066:150005] [ 4] sst(_ZN3SST5Event7executeEv+0x23)[0x90b43d]
[sb1066:150005] [ 5] sst(_ZN3SST10Simulation3runEv+0x23d)[0x969cc9]
[sb1066:150005] [ 6] sst[0x880741]
[sb1066:150005] [ 7] sst(main+0x1726)[0x881fd9]
[sb1066:150005] [ 8] /lib64/libc.so.6(__libc_start_main+0xfd)[0x2aaaaca2dd5d]
[sb1066:150005] [ 9] sst[0x87f6e9]
[sb1066:150005] *** End of error message ***

h4u5, 2016-04-29:
Looks like the seg faults are starting way before the timer runs out.

nmhamster, 2016-07-05:
@mjleven has fixed this bug.

closed 2016-07-05 by nmhamster