sstsimulator/sst-elements #304: Out of source build does not work

mjleven, 2016-05-17:
Out of source build does not work because of incorrect compiler -I argument.

/home/mjleven/sstGIT/split/elements/build/src/sst/elements/ember
[mjleven@sst-test ember]$ make
depbase=`echo embergen.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/sh ../../../../libtool  --tag=CXX   --mode=compile g++
-DHAVE_CONFIG_H -I. -I../../../../../src/sst/elements/ember
-I../../../../src
-I../../../../../src/sst/elements/ember/sirius/include
-I/usr/include/python2.7 -I/usr/include/python2.7
-I/usr/local/module-pkgs/boost/boost-1.56.0-nompi_gcc-4.8.3/include
-I/home/mjleven/sstGIT/split/core/build/install/include/sst/core
-I/home/mjleven/sstGIT/split/core/build/install/include  -g -O2 -g -O2
-std=c++1y -D__STDC_FORMAT_MACROS -MT embergen.lo -MD -MP -MF
$depbase.Tpo -c -o embergen.lo
../../../../../src/sst/elements/ember/embergen.cc &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  g++ -DHAVE_CONFIG_H -I.
-I../../../../../src/sst/elements/ember -I../../../../src
-I../../../../../src/sst/elements/ember/sirius/include
-I/usr/include/python2.7 -I/usr/include/python2.7
-I/usr/local/module-pkgs/boost/boost-1.56.0-nompi_gcc-4.8.3/include
-I/home/mjleven/sstGIT/split/core/build/install/include/sst/core
-I/home/mjleven/sstGIT/split/core/build/install/include -g -O2 -g -O2
-std=c++1y -D__STDC_FORMAT_MACROS -MT embergen.lo -MD -MP -MF
.deps/embergen.Tpo -c ../../../../../src/sst/elements/ember/embergen.cc
-fPIC -DPIC -o .libs/embergen.o
In file included from
../../../../../src/sst/elements/ember/embergen.cc:13:0:
../../../../../src/sst/elements/ember/emberengine.h:24:40: fatal error:
sst/elements/hermes/hermes.h: No such file or directory
#include <sst/elements/hermes/hermes.h>
                                        ^
compilation terminated.
make: *** [embergen.lo] Error 1

nmhamster, 2016-05-29:
Initial fix is committed and testing in PR #338.

jpvandy, 2016-06-03:
Nightly tests on sst-test and El Capitan are now passing

nmhamster, 2016-06-07:
This seems to be working successfully.

closed 2016-06-07 by nmhamster