sstsimulator/sst-elements #329: Make dist behavior with the new autotools file is unacceptable

jpvandy, 2016-05-24:
Previously, "make dist" following the Makefile.am instructions, wrote all directories except those with a .ignore to the tar file.  The new autogen is making decisions based on something else and vomiting directories for the "follow the Makefile.am" rule.

On the other  hand something is sending a configure.m4 file to the tar for each directory, including those with a .ignore.   Hence the directories exist, but some are otherwise unpopulated.

This is being seen on the elements repository.

(I have small, necessary, changes to the elements repo for make dist that are not yet in devel, but that do not address the above issues.)

jpvandy, 2016-06-01:
It would appear that the configure.m4 file in Ariel, chdl, HMC, Qsim and sysC have not been appropriately updated to the split core.   Clearly some kind of autotools mismatch.

allevin, 2016-06-07:
Fixed in PR #351

closed 2016-06-09 by allevin