sstsimulator/sst-elements #342: Firefly.nic gives a FATAL when running Ember

fkaplan3, 2016-06-01:
I get the following error when running Halo3D motif in ember (uses firefly.nic) on a dragonfly network:
FATAL: 9144874000:29:Nic::state_move_1():282 buffer not empty, 36 bytes remain

Use the following parameters to replicate the issue.
**Motif parameters:**
['Halo3D', 'iterations=50', 'doreduce=0', 'pex=17', 'pey=4', 'pez=4']
Halo3D problem size: 100x100x100
Halo3D iterms/cell: 1
numNodes=136
ranksPerNode=2

**Network parameters:**
topology=dragonfly2 shape=2:4:1:17
pktSize=1024B flitSize=8B
routingAlg=minimal
--host_bw=1GB/s --group_bw=1GB/s --global_bw=1GB/s --netBW=1GB/s

**Notes:**
It happens in both dragonfly and dragonfly2 networks.
Simulations using the same motif&network settings did not have this problem 2 weeks ago.

jwilso, 2016-06-29:
Arun:  Uncertain of this needs to be fixed by 6 or is a more subtle bug.

fkaplan3, 2016-06-29:
I think Michael already fixed it. Michael, could you confirm that?

Best,
Fulya Kaplan

closed 2016-07-01 by nmhamster