sstsimulator/sst-elements #353: configure in sst-elements 6.0.x

h4u5, 2016-06-07:
New Issue for sst-elements

`1` Detailed description of problem or enhancement
When running `./configure  --prefix=$SST_ELEMENTS_HOME --with-sst-core=$SST_CORE_HOME`
, there is some incompatibility with basename that prints the following bit of text out multiple times.

```
Configuration Information (Dependencies):

basename: invalid option -- 's'
Try `basename --help' for more information.
                                       : No
```

This does not impact the ability of the build.

`2` Describe how to reproduce
Following the build process on http://sst-simulator.org/SSTPages/SSTBuildAndInstall6dot0dot0SeriesDetailedBuildInstructions/

get to the sst-elements stage, then run configure as specified.

`3` What Operating system(s) and versions 
Red Hat Enterprise Linux Server release 6.7 (Santiago)
2.6.32-573.26.1.1chaos.ch5.4.x86_64

`4` What version of external libraries (Boost, MPI)
openmpi-1.8-gnu, boost_1_56_0

`5` Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)
This is from my fork of elements, but the remote is 29501290bb7e1a4e77c096e24ddac1c65d62865b

`6` Fill out Labels, Milestones, and Assignee fields as best possible

allevin, 2016-06-07:
Apparently -s option is not supported for basename on RHEL 6.6 and onward.  RHEL may be distributing an older version of basename.  I will try to figure out a workaround.

allevin, 2016-06-09:
Fixed via Pull Request #358

closed 2016-06-09 by allevin