sstsimulator/sst-elements #357: Configuration summary doesn't show correct result for goblin hmcsim

gvoskuilen, 2016-06-08:
When I configure the elements with '--goblin-hmcsim=', the configure shows it finds goblin:
checking hmc_sim.h usability... yes
checking hmc_sim.h presence... yes
checking for hmc_sim.h... yes
checking for hmcsim_clock in -lhmcsim... yes
checking for GOBLIN HMC Simulator library... yes

but in the printout of 'Configuration Information (Dependencies):', it shows:
goblin_hmcsim : No

This should report 'Yes'.

nmhamster, 2016-06-09:
@gvoskuilen you using GOBLINSim 1 or 2? I think this bug stands but I want to catalog which ones we know work as well. Thanks.

allevin, 2016-06-09:
The actual problem was introduced by the new configuration reporting changes.  The problem has been identified and corrected in PR #358.

gvoskuilen, 2016-06-10:
I was using GOBLINSim 2. I'm closing this since the bug is fixed.

closed 2016-06-10 by gvoskuilen