sstsimulator/sst-elements #369: Pymerlin.py Gives Segmentation Fault?

shavvn, 2016-06-13:
Hi,

I'm building new topologies for merlin routers, but as I tried the pymerlin test facility it gives segmentation fault, not only for my new topology, but also for the existing topologies too. 

I'm using the latest devel branch of sst-core and sst-elements. Here's how to reproduce the error, using torus as an example :

>  SST Core: #main() My rank is (0.0), on 1/1 nodes/threads
> SSTPythonModel Creating config graph for SST using Python model...
> Merlin SDL Generator
> 
> Please select network topology:
> [ 1 ]  Torus
> [ 2 ]  Fat Tree
> [ 3 ]  Dragonfly
> [ 4 ]  Simple
> [ 5 ]  Pentagon
> 1
> Please select endpoint:
> [ 1 ]  Test End Point
> [ 2 ]  Pattern-based traffic generator
> [ 3 ]  Bisection Test End Point
> 1
> Please enter num_dims:
> 2
> Dim 0 size:
> 3
> Dim 1 size:
> 3
> Dim 0 width (# of links in this dimension):
> 9
> Dim 1 width (# of links in this dimension):
> 9
> Please enter torus:local_ports:
> 1
> Please enter flit_size:
> 8B
> Please enter link_bw:
> 1GB/s
> Please enter xbar_bw:
> 10GB/s
> Please enter input_latency:
> 1ns
> Please enter output_latency:
> 1ns
> Please enter input_buf_size:
> 128B
> Please enter output_buf_size:
> 128B
> Please enter link_lat:
> 1ns
> SSTPythonModel Configuration will now parse parameters.
> SSTPythonModel Construction of config graph with Python is complete.
>  SST Core: # ------------------------------------------------------------
>  SST Core: # Graph construction took 39.724200 seconds.
>  SST Core: # Graph partitioning took 0.000105143 seconds.
>  SST Core: # Graph construction and partition raised RSS by 2432 KB
>  SST Core: # ------------------------------------------------------------
>  SST Core: Signal handers will be registed for USR1, USR2, INT and TERM...
>  SST Core: Signal handler registration is completed
> [:29054] *** Process received signal ***
> [:29054] Signal: Segmentation fault (11)
> [:29054] Signal code: Address not mapped (1)
> [:29054] Failing at address: 0x34436
> [1:29054] *** End of error message ***
> Segmentation fault

Any pointers of how to solve this problem?

Another unrelated issue: your developer email list doesn't seem to work... The link on your website http://sst-simulator.org/SSTPages/SSTMainSupport/ for developers doesn't work, and I sent an email a couple of days ago but it didn't show up in my inbox.

Thanks,
Shang

nmhamster, 2016-06-15:
@feldergast @bjmoor can you comment on this issue?

bjmoor, 2016-06-15:
@shavvn , can you run SST with GDB to get a backtrace to point out where exactly the fault is occuring?  I cannot replicate the segfault on my system.

shavvn, 2016-06-15:
@nmhamster @bjmoor  Thanks for the follow up!

I think I've located the problem: if I build merlin with my new topology then this would happen, even if I'm using torus here. So it looks like the python module didn't build successfully (though no error occurs during compilation). 

I'll try to get more details on this. But just to make sure of how to add a new topology, I only need to do the following:
 
- write the *new_topo.h* and *new_topo.cc* files
- add them to *Makefile.am*
- add an entry in *merlin.cc*
- add an entry in *boostExport.cc*
- write corresponding python class in *pymerlin.py*

Is that correct?

Thanks!

shavvn, 2016-06-17:
After working around I figured out this problem: one of my class method wasn't implemented correctly but somehow it escaped the compilation process... It's a bug in my code so I'm closing this issue. Sorry for bugging you about this.

closed 2016-06-17 by shavvn