sstsimulator/sst-elements #404: Trac: #400 Configure check for PIN should abort if too old of a version is found

allevin, 2016-06-29:
When I .configure using --with-pin=${INTEL_PIN_DIRECTORY} and the directory which I point to contains a version of PIN which is too old to work with the current SST, an error should occur, rather than blind acceptance. It is not nice to have configure work, but then make fail with a very non-obvious compilation error (missing header files).

NOTE This was issue [11 in sst-core](https://github.com/sstsimulator/sst-core/issues/11) and has been transferred to sst-elements.

nmhamster, 2016-07-08:
We will look to add this in the PIN overhaul for version 3.0 during SST v6.1.0 development.

allevin, 2020-04-03:
@gvoskuilen Is this something that should be looked at for PIN3 support?

closed 2020-09-30 by allevin