sstsimulator/sst-elements #414: Ariel build-system / pintool issues

bjmoor, 2016-07-07:
Current pintool build system requires PIN 2.14.   Should support 3.0 as well.
Currently cannot use C++11 features in fesimple, because C++11 flags aren't there.

nmhamster, 2016-07-07:
Not supported yet, we started work on this for SST v6.0.0 but the new "mimic OS" features PIN have added create a lot of mess with the build system. I don't think we will officially support PIN-3.0 until SST 6.1 or 7.0.

dskarlatos, 2017-06-16:
@nmhamster  is there an update on this? Thanks!

gvoskuilen, 2020-04-01:
Pin3 is now supported

closed 2020-04-01 by gvoskuilen

sheridp, 2021-02-04:
> Pin3 is now supported

When running `./configure --prefix=/opt/sst-elem --with-sst-core=/opt/sst --with-pin=~/pin-3.17-98314-g0c048d619-gcc-linux` I am getting 
```
checking for pin.sh... no
checking for pin... no
checking PIN Tool... no
configure: error: PIN was requested but not found
```
Am I doing something wrong to use pin3?

allevin, 2021-02-05:
@sheridp This issue is closed.  However, check that the file "pin" exists in the ~/pin-3.17-98314-g0c048d619-gcc-linux directory. You can see how SST detects for pin via this file https://github.com/sstsimulator/sst-elements/blob/master/config/sst_check_pintool.m4 

If you continue to have problems, please open a new issue.

sheridp, 2021-02-19:
Hi allevin, thanks for your response.  It turns out this issue was a[ peculiar shell issue ](https://unix.stackexchange.com/questions/373519/expansion-of-tilde-in-zsh), such that my `--with-pin=~/pin-3.17-98314-g0c048d619-gcc-linux` was not being expanded.