sstsimulator/sst-elements #433: Stats warningcould be slightly more helpful

cdunham, 2016-08-24:
New Issue for sst-elements

`1` Detailed description of problem or enhancement

`2` Describe how to reproduce

`3` What Operating system(s) and versions 

`4` What version of external libraries (Boost, MPI)

`5` Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

`6` Fill out Labels, Milestones, and Assignee fields as best possible

allevin, 2016-08-24:
Curtis,

Can you provide a better description of the problem?  Is this an issue with the Statistics engine in the core?  Is is a problem with sst-info?  or is it a problem with MemHierarchy?

What is the problem with the warning, and what do you suggest as a possible solution?

cdunham, 2016-08-24:
Simply s/sstinfo/sst-info/ the messages as that seems to be the name of the
tool when built from sst 6.0 tarballs.

nmhamster, 2016-08-24:
@allevin @gvoskuilen see PR #434 for fix.

allevin, 2017-03-24:
Not enough information to resolve this issue.  PR#434 shows an attempted fix.

closed 2017-03-24 by allevin