sstsimulator/sst-elements #463: Add SST Statistic string type

jleidel, 2016-10-10:
The current SST::Statistics class infrastructure currently supports numeric types.  There are a number of external modules/components that may require raw logging capabilities.  The HMC-Sim simulator is one such component.  The issue being that individual HMC-Sim logs may contain multiple integer components.  For example, tracking long latency events (cross-chip routing) requires at least two numerical components (source link, destination quad).  Tracking individual requests within HMC-Sim also requires string output as the current HMC-Sim has arbitrary CMC processing capability.    

label:Enhancement

nmhamster, 2016-10-10:
@jleidel would it be helpful to support a "custom" class/struct for tracking multiple variables?

jleidel, 2016-10-10:
@nmhamster , the tracing in HMC-Sim is somewhat hierarchical, so we need the ability to trace the intersection of up to 7 variables.  EG: 
* device id
* device quad
* device vault
* source device (CUB)
* destination device (CUB)
* device link 
* queue slot
* request type <string>

If we can accomplish such a feat with an external class/struct, that would be great. Possibly an SST:Statistics template?

jleidel, 2018-01-12:
This has since been updated to utilize the latest tracing enhancements.  No longer a feature request.

closed 2018-01-12 by jleidel