sstsimulator/sst-elements #467: CramSim appears to not be thread-safe

jpvandy, 2016-10-12:
This is not a solid failure. The manifestion is one or more output lines mixed with each other.  It is seen less than once per execution of the 11 test Suite.  It is not the same test each time.

A sample output despondency is 

! CramSim_6_W test Fails
! shunit.ASSERT: outFile word/line count does NOT matches Reference
! 5,6d4
! < Deleting DIMM
! < Refresh's sent out: 0
! 11c9,11
! < Total Txns Received: 24776
! ---
! > Deleting DIMMTotal Txns Received: 24776
! > 
! > Refresh's sent out: 0

The use of std:cout is probably the source of the problem.
The code in the directory .../elements/CramSim has 514 references to cout.

jpvandy, 2016-11-09:
The CramSim testing criterion was significantly reduce to ignore the wrongly printed output when running multithread.   It is believed that the calculation is correct only the printed output is wrong.

gvoskuilen, 2020-10-15:
Addressed in #1545

closed 2020-10-15 by gvoskuilen