sstsimulator/sst-elements #469: Simulation configuration file with Ariel and memHierarchy

dhiraj113, 2016-10-14:
Is there an example simulation configuration file (sdl.py) with the components ariel and memHierarchy connected together? 

I haven't found even a single configuration file with ariel and memHierarchy components connected. Also, in general, what is the best approach to figuring out how to write configuration files for connecting various components?

gvoskuilen, 2016-10-14:
There's a couple places you can look. On the sst-simulator.org site, if you go to Downloads, and find the "tutorials" section, there are examples for connecting both Ariel & Miranda to memHierarchy in "Building a Simulator Examples". The other place you can look is in src/sst/elements/ariel/frontend/simple/examples/stream/.

nmhamster, 2016-12-07:
@dhiraj113 do you have what you need for your question? If so, we can close the issue. If you don't let us know so we can help some more. Thanks!

dhiraj113, 2016-12-08:
I have been using the ariel_snb.py configuration file. It solves my requirements for now. So, I am closing the issue. Thanks.

closed 2016-12-08 by dhiraj113