sstsimulator/sst-elements #470: fork: Cannot allocate memory

dhiraj113, 2016-10-14:
New Issue for sst-elements

`1` Detailed description of problem or enhancement
Fatal error in Ariel component because of failure to allocate memory. I am running the code in a virtual machine with 12GB of RAM.

```
fork: Cannot allocate memory
FATAL: ArielComponent[arielcpu.cc:392:forkPINChild] Fork failed to launch the traced process. errno = 12
FATAL: ArielComponent[arielcpu.cc:392:forkPINChild] Fork failed to launch the traced process. errno = 12
```

`2` Describe how to reproduce

` sst sst-elements/src/sst/elements/ariel/frontend/simple/examples/stream/ariel_snb.py`

`3` What Operating system(s) and versions 

```
Ubuntu 14.04.5 LTS
Linux kernel version - 3.19.0-69-generic
GCC version - 4.8.4
```

`4` What version of external libraries (Boost, MPI)

```
Boost version - 1.56
MPI version - 1.8
```

`5` Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)
sst 6.0.0 release versions with no modifications.

`6` Fill out Labels, Milestones, and Assignee fields as best possible

gvoskuilen, 2016-10-14:
Can you try adding the parameter:
"do_not_back" : 1
to mem_params in ariel_snb.py? 
This parameter tells the simulation not to allocate a backing store for the memory. That may be using up all the memory and causing fork to fail.

dhiraj113, 2016-10-14:
Thanks that fixes the issue.

nmhamster, 2016-10-17:
Closing issue as this seems resolved. If this continues please reopen the issue. Thanks for your help everyone!

closed 2016-10-17 by nmhamster