sstsimulator/sst-elements #522: Generate configure file

dhiraj113, 2016-11-28:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement
How to generate the configure file required to build the code in the git repository. Why is that the release versions have a configure file whereas the git repository doesn't?

I have tried running autoconf to generate the configure file, I get the following error 

`/usr/bin/m4:configure.ac:20: cannot open `config/sst_elements_include.m4': No such file or directory
autom4te: /usr/bin/m4 failed with exit status: 1
`

2 - Describe how to reproduce
autoconf

3 - What Operating system(s) and versions 
Ubuntu 14.40

4 - What version of external libraries (Boost, MPI)

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

6 - Fill out Labels, Milestones, and Assignee fields as best possible

dhiraj113, 2016-11-28:
Run ./autogen.sh to generate the configure file

If you get the below error on running ./autogen.sh, install libtool

`Unable to find working libtool. [][]`

closed 2016-11-28 by dhiraj113

jwilso, 2016-11-28:
In the SST [detailed build instructions](http://sst-simulator.org/SSTPages/SSTBuildAndInstall6dot0dot0SeriesDetailedBuildInstructions/#WhatsNext), it states the following:

>To clone the latest project source code on the devel branch (from GitHub):
    Using Git
        git clone -b devel https://github.com/sstsimulator/sst-core.git
        git clone -b devel https://github.com/sstsimulator/sst-elements.git
    The build process is the same as described above, except the user must perform a $ ./autogen.sh to create a configure file. This must occur before running the configure script.