sstsimulator/sst-elements #538: Ember: Sweep3D failing

h4u5, 2016-12-13:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement
Ember Sweep3D is failing. with std::out_of_range (stoi)

2 - Describe how to reproduce
EMBER: network: topology=dragonfly2 shape=24:48:12:96 arbitration=xbar_arb_lru, routing=minimal
EMBER: network: topology=dragonfly2 shape=24:48:12:96 arbitration=xbar_arb_lru
EMBER: numNodes=110592 numNics=110592
EMBER: network: BW=12.5GB/s pktSize=2048B flitSize=32B
EMBER: Job: -nidList=0-110591 -ranksPerNode=1 ['Init']
EMBER: Job: -nidList=0-110591 -ranksPerNode=1 ['Sweep3D', 'nx=100', 'ny=100', 'nz=100', 'pex=384', 'pey=288', 'kba=10', 'nodeflops=5000000000000.000000', 'fields_per_cell=6']
EMBER: Job: -nidList=0-110591 -ranksPerNode=1 ['Fini']
Merlin parameters: (key, value)
dragonfly:algorithm, minimal
dragonfly:intergroup_per_router, 24
num_peers, 110592
output_buf_size, 28KB
router_radix, 95
num_vns, 1
dragonfly:routers_per_group, 48
num_ports, 95
link_lat, 150ns
link_bw, 12.5GB/s
dragonfly:global_route_mode, absolute
dragonfly:intergroup_links, 12
flit_size, 32B
topology, merlin.dragonfly2
dragonfly:hosts_per_router, 24
output_latency, 150ns
xbar_bw, 12.5GB/s
dragonfly:num_groups, 96
xbar_arb, merlin.xbar_arb_lru
input_buf_size, 28KB
input_latency, 150ns
dragonfly:shape,
debug, 0
Round robin partitioning
terminate called after throwing an instance of 'std::out_of_range'
  what():  stoi

3 - What Operating system(s) and versions 
Skybridge

4 - What version of external libraries (Boost, MPI)
module load openmpi-gnu/1.8
boost_1_56_0

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)
sst-elements: 4108adb51241b77ec6f8b47a7f6c20253288e049

6 - Fill out Labels, Milestones, and Assignee fields as best possible

h4u5, 2017-01-31:
Just an update that I tried to run at a smaller scale of 9984 nodes and it still failed with a similar message

hughes-c, 2020-01-26:
@h4u5 @mjleven Is this error still present?

closed 2020-10-07 by allevin