sstsimulator/sst-elements #548: SST on MacOS Sierra is not working with PIN

jpvandy, 2017-01-06:
From Prospero.pin the failure message is:

## --------------------------- run the pin trace program
/Users/sstbuild/jenkins/workspace/SST__Nightly_MacOSX_10.12_OMPI_1.8_Boost_1.56_clang_800.0.42.1_mainline/24_24/local/sst-elements/bin/sst-prospero-trace
A: Source/pin/injector_m/vm_mac_loader.cpp: GenerateMiniloaderProgramForImageLoad: 494: preferredMhAddr not found

NO STACK TRACE AVAILABLE
Detach Service Count: 1
Pin 2.14
Copyright (c) 2003-2015, Intel Corporation. All rights reserved.
@CHARM-VERSION: $Rev: 71293 $
@CHARM-BUILDER: BUILDER
@CHARM-COMPILER: gcc 4.2.1
@CHARM-TARGET: ia32
@CHARM-CFLAGS:  __OPTIMIZE__=1  __NO_INLINE__=__NO_INLINE__

From Ariel test the fail message is:
test_Ariel_runstream
 starting Directory /Users/sstbuild/jenkins/workspace/SST__Nightly_MacOSX_10.12_OMPI_1.8_Boost_1.56_clang_800.0.42.1_mainline/24_24/devel/trunk
/Users/sstbuild/jenkins/workspace/SST__Nightly_MacOSX_10.12_OMPI_1.8_Boost_1.56_clang_800.0.42.1_mainline/24_24/devel/trunk/sst-elements/src/sst/elements/ariel/frontend/simple/examples/stream/runstream.py
FATAL: ArielComponent[arielcpu.cc:466:forkPINChild] Launching trace child failed!  Child Terminated With Signal 6; Core Dump File Created = 0
 
WARNING: sst did not finish normally, RetVal= 1

allevin, 2020-04-03:
@jpvandy is this still an issue?

jpvandy, 2020-04-03:
working on Mojave

closed 2020-04-03 by jpvandy