sstsimulator/sst-elements #625: Ariel pin with Ubuntu 16.04 LTS (64-bit) with GCC v5.4.0

dhiraj113, 2017-04-02:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement
The SST release page mentions Ubuntu 16.04 LTS (64-bit) with GCC v5.4.0 as one of the OS/Compiler combinations for which sst has been tested. I am curious to know how you have got sst-elements to build with pin-tool (to enable ariel) while using gcc 5.4. When I try to build sst with pin using gcc version 5.4, it produces the following error showing the incompatability of the ABI

```
/opt/pin-2.14/source/include/pin/compiler_version_check2.H:102:2: error: #error The C++ ABI of your compiler does not match the ABI of the pin kit.

error: #error This kit requires gcc 3.4 or later
```

Also, pin tool 2.14 doesn't even run in isolation on linux kernel versions above 4.0. Trying to execute

./pin-2.14 on Ubuntu 16.04 with linux kernel version 4.4.0,  results in the error 

`E: 4.4 is not a supported linux release`

So, if the pin-tool used by ariel doesn't even run with linux kernel version present in Ubuntu 16.04 and the sst-elements doesn't even build ariel with gcc version 5.4, how is that the sst-elements has been successfully tested on the OS/compiler combination of Ubuntu 16.04 LTS (64-bit) with GCC v5.4.0?

2 - Describe how to reproduce
Try building sst-elements with ariel.
```
# with gcc 5.4
./configure --with-sst=<sst-core> --with-pin=<pin-2.14>
make
```

Try executing pin-2.14 in isolation on a linux kernel version above 4
```
# On a linux with kernel version > 4
./pin-2.14
```

3 - What Operating system(s) and versions 
Ubuntu 16.04 ( Kernel version - 4.4.0-71-generic)

4 - What version of external libraries (Boost, MPI)

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)
sst 6.1.0 release

6 - Fill out Labels, Milestones, and Assignee fields as best possible

amroawad2, 2017-04-04:
Hi Dhiraj,

Thanks for bringing this up. I followed up with the testing team and figured out the issue. As you could observe in the main web page of SST, Pin-2.14 is listed as an **Optional** external package. Apparently, our tests targeted the key components that have no external **_optional_** dependencies. That said, we are just preparing a list of the tested external dependencies and which tested environment had them working correctly.

dhiraj113, 2017-04-04:
Hi Amro,

Thanks for the update. I guess ariel compiles only with gcc versions < 5 and runs only on linux kernel versions < 4. (Both dependencies caused because of pin-2.14). While this is not a deal breaker, its a pain to work with a downgraded kernel on the development machine.

BTW, Besides Ariel, do you know of any other ways with which an arbitrary binary could be run with SST?

amroawad2, 2017-04-04:
We have few CPU models that are still development, e.g., RISC-V model, and other synthetic core models, e..g, Miranda, however, I suggest you stick with Pin for now. Meanwhile, please let us know if you have any questions.

nmhamster, 2017-04-07:
@dhiraj113 @amroawad2 we think there is the possibility of a fix for Ariel on Ubuntu 16.04 coming for the SST 7.0 release. We can get the new code to build successfully but there are some issues we are having to iron out. Given the release is one a few weeks away I am hopeful you won't have to wait /too/ long to have what you need.

dhiraj113, 2017-04-07:
@nmhamster Thanks for the update. Looking forward to the SST 7.0 release!

dhiraj113, 2017-04-13:
Hi Hammond,

Mimiking your changes in the branch ariel-pin32812, I have also got ariel to build on Ubuntu 16.04 with gcc 5.4. However, I am not able to run ariel. I am curious to know if you have got to run an ariel test case correctly as well?

I get the below error when running a testcase with a ariel component.

```
sst src/sst/elements/ariel/frontend/simple/examples/stream/runstream.py

E: Unable to load /opt/sst-elibexec/fesimple.so: dlopen failed: library "libdl.so.2" not found
FATAL: ArielComponent[arielcpu.cc:461:forkPINChild] Launching trace child failed!  Child Exited with status 255
FATAL: ArielComponent[arielcpu.cc:461:forkPINChild] Launching trace child failed!  Child Exited with status 255
```

Adding libdl.so.2 to LD_LIBRARY_PATH doesn't seem to help either.

nmhamster, 2017-04-13:
Hi @dhiraj113 - we are working through these errors. Intel has made a significant change in PIN-3 series so that it provides its own internal copies of the C++ STL and decides how to load libraries. It elects to now pass through `LD_LIBRARY_PATH` information and so you get errors like this. I am hoping to have something to give to you shortly, we can get through some of the errors you have above by using static libraries but it does not completely remove all the problems we have found yet. I am sorry this is taking so long and keeping you from your work.

dhiraj113, 2017-04-13:
@nmhamster That explains the libdl.so not found error. Thanks again for looking into this issue. It's not really stopping my work. I am getting around by running with a downgraded kernel. The only problem is that I can't get sound to work on my motherboard with the downgraded kernel :).

dhiraj113, 2017-04-15:
Found a similar discussion thread regarding PIN 3.0 on zsim git repo. Including it here

https://github.com/s5z/zsim/issues/109

yarcod, 2017-05-17:
Having exactly the same problem as OP - have you been able to get everything to work with Pin-3 in the 7.0.0 release? :)

dskarlatos, 2017-06-12:
@nmhamster is there an update on this? Thanks!

hughes-c, 2017-06-16:
@dskarlatos I don't believe that we have an ETA on getting Pin 3 integrated with SST. The library issues described by @nmhamster's comment above is the reason for the delay.

dskarlatos, 2017-06-16:
@hughes-c OK, thank you for the update.

closed 2017-11-06 by dhiraj113