sstsimulator/sst-elements #655: EmberSweep encounters occasional time limits

jpvandy, 2017-05-12:
I observe that the changes that went into Release 7.0 have significantly reduced the probability of Ember Sweep time limit.   The nightly report has caught two this month.   I am going close issue #274 on this subject.  But I will open this new Issue to keep track of the much less frequent failures.     EmberSweep consists of 180 tests.  The Suite is run more than a dozen times in the nightly.  Two events seen in half a month is relatively low probability.    
May 2nd, test #58 experience time limit on COE-RHEL 6 and on May 11th, there was a time limit on test #57 on El Capitan X-code-8.    Unfortunately neither of these builders support Branden's recently added trace-back code.

jpvandy, 2017-05-18:
Time limit on test 80 on El Capitan, number of ranks=2,  run #445.
This resulted in a hang, which I believe is a SQE problem, not an Ember problem.

jpvandy, 2017-06-19:
Time limit on test 170 on COE_RHEL 6 on June 18th.  May have been others since the May 18 event.  Trace-back is unavailable, but deadlock test failed and that needs to be examined.

bjmoor, 2017-06-19:
The EmberSweep_170 on June 18th cannot be deadlock, as that test was not run as a parallel simulation, correct?  We need to get trace-back information to have any chance of detecting if the time-limit was due to the SST behaving poorly, or if it was external factors.

closed 2020-10-14 by allevin