sstsimulator/sst-elements #671: Segmentation fault when using Ariel

janzhou, 2017-05-26:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

SST runs into a segmentation fault. The application run natively without issues. I have tried both 7.0.0 and devel.

```
ARIEL: Enabling memory and instruction tracing from program control.
[ubuntu-14:02168] *** Process received signal ***
[ubuntu-14:02168] Signal: Segmentation fault (11)
[ubuntu-14:02168] Signal code: Address not mapped (1)
[ubuntu-14:02168] Failing at address: 0x28
[ubuntu-14:02168] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x10330) [0x7fbcd1490330]
[ubuntu-14:02168] [ 1] /home/jan/local/sstelements-7.0.0/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy21MemHierarchyInterface11sendRequestEPNS_10Interfaces9SimpleMem7RequestE+0xa1) [0x7fbcc85a0d11]
[ubuntu-14:02168] [ 2] /home/jan/local/sstelements-7.0.0/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent9ArielCore15commitReadEventEmmj+0xf8) [0x7fbcc8b303c8]
[ubuntu-14:02168] [ 3] /home/jan/local/sstelements-7.0.0/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent9ArielCore17handleReadRequestEPNS0_14ArielReadEventE+0x7d) [0x7fbcc8b3049d]
[ubuntu-14:02168] [ 4] /home/jan/local/sstelements-7.0.0/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent9ArielCore16processNextEventEv+0x3d8) [0x7fbcc8b31288]
[ubuntu-14:02168] [ 5] /home/jan/local/sstelements-7.0.0/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent9ArielCore4tickEv+0x36) [0x7fbcc8b31736]
[ubuntu-14:02168] [ 6] /home/jan/local/sstelements-7.0.0/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent8ArielCPU4tickEm+0x88) [0x7fbcc8b25c78]
[ubuntu-14:02168] [ 7] sst(_ZN3SST5Clock7executeEv+0xee) [0x46e6be]
[ubuntu-14:02168] [ 8] sst(_ZN3SST10Simulation3runEv+0x1c7) [0x4b9997]
[ubuntu-14:02168] [ 9] sst() [0x45fbba]
[ubuntu-14:02168] [10] sst(main+0xf0b) [0x4538fb]
[ubuntu-14:02168] [11] /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7fbccf99df45]
[ubuntu-14:02168] [12] sst() [0x45f5e1]
[ubuntu-14:02168] *** End of error message ***
Segmentation fault
```

GDB infos I get:

```gdb
Program received signal SIGSEGV, Segmentation fault.
0x00007fffeead2d11 in send (event=0x7fffeddd7dc8, delay=0, this=0x0) at /home/jan/local/sstcore-7.0.0/include/sst/core/link.h:89
89              send(delay,defaultTimeBase,event);
(gdb) where
#0  0x00007fffeead2d11 in send (event=0x7fffeddd7dc8, delay=0, this=0x0) at /home/jan/local/sstcore-7.0.0/include/sst/core/link.h:89
#1  send (event=0x7fffeddd7dc8, this=0x0) at /home/jan/local/sstcore-7.0.0/include/sst/core/link.h:96
#2  SST::MemHierarchy::MemHierarchyInterface::sendRequest (this=0x30b2550, req=0x104f17b0) at memHierarchyInterface.cc:46
#3  0x00007fffef0623c8 in SST::ArielComponent::ArielCore::commitReadEvent (this=this@entry=0x3211430, address=address@entry=356272,
    virtAddress=virtAddress@entry=140736882921392, length=length@entry=8) at arielcore.cc:139
#4  0x00007fffef06249d in SST::ArielComponent::ArielCore::handleReadRequest (this=this@entry=0x3211430, rEv=<optimized out>) at arielcore.cc:411
#5  0x00007fffef063288 in SST::ArielComponent::ArielCore::processNextEvent (this=this@entry=0x3211430) at arielcore.cc:578
#6  0x00007fffef063736 in SST::ArielComponent::ArielCore::tick (this=0x3211430) at arielcore.cc:659
#7  0x00007fffef057c78 in SST::ArielComponent::ArielCPU::tick (this=0xac24b0, cycle=<optimized out>) at arielcpu.cc:554
#8  0x000000000046e6be in SST::Clock::execute (this=0x7fffee5e2f40) at clock.cc:92
#9  0x00000000004b9997 in SST::Simulation::run (this=this@entry=0xa514a0) at simulation.cc:530
#10 0x000000000045fbba in start_simulation (tid=tid@entry=0, info=..., barrier=...) at main.cc:330
#11 0x00000000004538fb in main (argc=5, argv=0x7fffffffe458) at main.cc:687
(gdb) list
84            timebase.
85            @param delay The additional delay, in units of the default Link timebase
86            @param event The event to send
87            */
88          inline void send( SimTime_t delay, Event* event ) {
89              send(delay,defaultTimeBase,event);
90          }
91
92          /** Send an event with the Link's default delay
93            @param event The event to send
```

2 - Describe how to reproduce

a) checkout https://github.com/janzhou/missteps
b) compile: make
3) run: sst ariel.py thread_read.exe 8 128

The 3rd parameter 8 is the number of banks. Whenever the number of banks is equal or greater than 8, SST will throw a segmentation fault.

3 - What Operating system(s) and versions 

Ubuntu 14.04.5 LTS 64bit

4 - What version of external libraries (Boost, MPI)

SST-Ariel

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

6 - Fill out Labels, Milestones, and Assignee fields as best possible

nmhamster, 2017-05-26:
@janzhou @amroawad2 - is it possible to get the example code emailed into us to run and debug?

nmhamster, 2017-05-26:
@janzhou - is the missteps repo the code you are running, just want to check?

amroawad2, 2017-05-26:
Yes, I believe it is the missteps. Lets wait until Jian confirms

janzhou, 2017-05-26:
@nmhamster @amroawad2 Yes, the missteps repo is the code I am running.

janzhou, 2017-05-27:
I have moved from pthread to openmp. This issue still occurs.

amroawad2, 2017-05-27:
It is likely an issue from the configuration script having some cores not connected, and not from SST. It seems somehow Pthread or OpenMP are getting scheduled into the extra cores that are replaced by MMU, thus it is better to set the number of Ariel cores exactly to the actual ones getting connected to cache links.

amroawad2, 2017-05-27:
The issue is fixed 👍

closed 2017-05-27 by amroawad2