sstsimulator/sst-elements #674: MemHSieve occasional time limits

gvoskuilen, 2017-06-02:
MemHSieve encounters occasional timelimits during testing, probably because of a seg fault in fesimple/app.

El Capitan error output:
C: Tool (or Pin) caused signal 11 at PC 0x600105460
Pin child/app terminated abnormally due to signal 11.

Time limits also occur on CentOS but there is no error output.

jpvandy, 2017-06-03:
Believe the missing pin/child is ompsievetest

jpvandy, 2017-06-19:
When the time limit is detected, Ariel is in the process of replacing malloc with Ariel equivalent.  The time limit is reported after three of the six lines, "Identified routine ... replacing with Ariel equivalent" have been written.

Here are the initial lines of output from a successful memHSieve test run:

test_memHSieve 
done configuring SST
ArielComponent[arielcpu.cc:52:ArielCPU] Creating Ariel component...
ArielComponent[arielcpu.cc:58:ArielCPU] Configuring for 16 cores...
ArielComponent[arielcpu.cc:61:ArielCPU] Configuring for check addresses = no
ArielComponent[arielcpu.cc:143:ArielCPU] Loading memory manger: ariel.MemoryManagerSimple
ArielComponent[arielcpu.cc:151:ArielCPU] Memory manager construction is completed.
ArielComponent[arielcpu.cc:182:ArielCPU] Model specifies that there are 0 application arguments
ArielComponent[arielcpu.cc:192:ArielCPU] Interception and instrumentation of multi-level memory calls is ENABLED.
ArielComponent[arielcpu.cc:197:ArielCPU] Tracking the stack and dumping on malloc calls is ENABLED.
ArielComponent[arielcpu.cc:202:ArielCPU] Base pipe name: /sst_shmem_10796-0-1681692777
ArielComponent[arielcpu.cc:213:ArielCPU] Processing application arguments...
ArielComponent[arielcpu.cc:316:ArielCPU] Completed processing application arguments.
ArielComponent[arielcpu.cc:323:ArielCPU] Creating core to cache links...
ArielComponent[arielcpu.cc:327:ArielCPU] Creating core to allocate tracker links...
ArielComponent[arielcpu.cc:333:ArielCPU] Creating processor cores and cache links...
ArielComponent[arielcpu.cc:336:ArielCPU] Configuring cores and cache links...
ArielComponent[arielcpu.cc:362:ArielCPU] Registering ArielCPU clock at 2660 MHz
ArielComponent[arielcpu.cc:366:ArielCPU] Clocks registered.
ArielComponent[arielcpu.cc:374:ArielCPU] Completed initialization of the Ariel CPU.
SSTARIEL: Loading Ariel Tool to connect to SST on pipe: /sst_shmem_10796-0-1681692777 max instruction count: 1000000000 max core count: 16
SSTARIEL: Function profiling is disabled.
ARIEL-SST: Did not find ARIEL_OVERRIDE_POOL in the environment, no override applies.
ARIEL-SST PIN tool activating with 16 threads
ARIEL: Default memory pool set to 0
ARIEL: Tool is configured to begin with profiling immediately.
ARIEL: Starting program.
Identified routine: malloc/_malloc, replacing with Ariel equivalent...
Identified routine: malloc/_malloc, replacing with Ariel equivalent...
Identified routine: free/_free, replacing with Ariel equivalent...
Identified routine: malloc/_malloc, replacing with Ariel equivalent...
Identified routine: free/_free, replacing with Ariel equivalent...
Identified routine: clock_gettime, replacing with Ariel equivalent...
Replacement complete.

closed 2020-10-14 by allevin