sstsimulator/sst-elements #680: Merlin Topologies

Christian9212, 2017-06-13:
Detailed description of problem or enhancement
Hi everyone, i have recently installed SST, and i am strugling a bit with "know-how"/"tutorials"/"guidelines" regarding network topologies, and full-system simulation. I can barely find any examples out there. What i really need right now is an Fat-tree network topology, but dont know exactly how to do it. Even so, i am still trying and learning to use the SST.

Any relevant info please? And sorry for any mistakes, but i'm new to SST and i would like to use it.

 What Operating system(s) and versions 
Linux-Ubuntu 16

Thx

nmhamster, 2017-06-16:
Hi @Christian9212 - here is a response from one of our developers on this question recently.

>    The simplest way to generate a topology (including Fat Tree) with Merlin is to use the built-in topology generators. Manually setting up something like a large fattree network is definitely error prone (but not impossible).
>  
>   Please have a look at https://github.com/sstsimulator/sst-elements/blob/devel/src/sst/elements/merlin/networkGen.py
>  
>    It gives you the option of choosing your topology, and your endpoint type, and prompts you to supply the configuration parameters that you want to use for the simulation.

Let us know if this helps you get started or if you need any additional information. Thanks for your interest in SST!

allevin, 2020-04-03:
No response from author, considering this issue closed

closed 2020-04-03 by allevin